portapack-mayhem/mayhem-firmware #2152: Boot taking long time (30s to 1min)

rafaelmoco, 2024-05-17:
### What happened?

This is my second portapack. The first one stopped to boot every time I plug an SD card. It can boot without the card, and I already tried three different SDs

So I moved on and bought a new portapack, a Mayhem Edition this time, using the link from the official Wiki. Everything is back to normal, BUT sometimes the boot screen takes a long time to load, something similar to what was happening to the other portapack before it stops to boot with an SD connected

The SD card is a Sandisk with 32GB, Fat32 formatted on Linux, running Mayhem 2.0.1 

OBS: The first Portapack stopped working using another SD Card, a 64GB Sandisk. This 32GB Sandisk is brand new and original

### Your hardware

H2M (Mayhem edition)

### Tests

- [X] I managed to update/boot previously

rafaelmoco, 2024-05-17:
btw, this is the boot screen I'm talking about, before the Splash screen. Sometimes my Portapack just stay locked in this screen, the new one is stuck even without the SDCard now
[image]

rafaelmoco, 2024-05-18:
Update: I've managed to boot and disable the Splash Screen. Seems to be booting fine now. I'll test some more, but maybe a bug with the splash screen loading?

gullradriel, 2024-05-31:
It's more plausibly a bug with your sdcard, or sdcard reader, or the data is damaged in the bitmap.
It's known that a damaged splash boot picture or in the bad format can freeze the boot.
Happy to see you found a way.

closed 2024-05-31 by gullradriel