portapack-mayhem/mayhem-firmware #2156: WarDriveMap - Support for Flipper's GPS Geotag Data (Sub-Driving / War-Driving)

neednotapply, 2024-05-21:
### Description of the feature you're suggesting.

A Flipper with a GPS module can capture GPS data and store it with it's .SUB file, however this data is not easily viewed on a map, just a lat/long.

My request is that the WarDriveMap app could support Flipper's SD file structure, the .SUB GEO data and Flipper's WarDriving GEO data.

### Anything else?

I realize that the two devices' capture formats are not compatible and it would not be practical to alter either, rather that the Flipper's SD could be placed into the PP. (or even just use a singular SD for both)

htotoo, 2024-05-21:
If there won't be a big NO from somebody I would implement this I think. Sooner or later.

But there is a marker limit in the map, because of the ram. So if you have some geotagged files, it won't show all.

This should be also solved, maybe with a paginator like option.

Ponohka, 2024-08-22:
У меня при включении этого режима GPS иногда появляются подделанные сигналы спутников GPS. Но практически всегда происходит глушение GPS сигналов. О том, что при включении этого режима происходит излучение на частоте 1575,42 МГц. это я фиксирую на анализаторе спектра. Кто подскажет в чем тут дело?

closed 2024-09-16 by htotoo

Ponohka, 2024-09-20:
Please tell me. When I launch the GPSSim function, I detect radiation at a frequency of 1575.42 MHz with a spectrum analyzer. But when I enable this function using the same fake coordinate file on my phone, it either fixes or doesn’t fix. What can I do to make this satellite simulator feature work all the time when turned on?

htotoo, 2024-09-20:
Hello. This app in the PR is note related to your question, this is for wardrive, not gps TX!

When you wanna use gpssim, try to check the wiki page: https://github.com/portapack-mayhem/mayhem-firmware/wiki/GPS-Sim
There is a lot of info about how to use it, and what are the difficulties.

Ponohka, 2024-09-21:
Спасибо