portapack-mayhem/mayhem-firmware #2157: TX MIC app - no audio

aneitzel, 2024-05-21:
### Describe the bug.

I used hackrf + portapack (MIC app) in the past for test transmissions which worked fine. Have Portapack with internal battery. Since i played around with mayhem v2.0 i now noticed, that there is no audio anymore then TX via MIC app. All other apps like soundboard etc. works fine. Also all other SDR stuff in HackRF mode.

> Then i open MIC app and talk into connected MIC , the audio level vumeter -in the left part- is not moving at all

It definitively worked in the past / the only thing i remember was palcing around with latest mayhem, and nightly builds.
I also tested to flash back to a old version, no differenies... now back at latest nightly 24521.

Any idea what to check? My MICs i connected working on other devices (this was my 1. assumption).

### Reproduction

switch on
open TX mic app (FMn, Frequency for example PMR K1, low pwr) 
connect mic to headphone jack
activate "hear mic" > no VU meter activity at all
Also during TX, no audio via mic / HF signal unmodulated in test receiver
---------------------------------
TX via soundboard app, audio is fine.

### Expected behavior

level VU meter activity / audio from mic expected during TX

gullradriel, 2024-05-21:
If going back isn't fixing it, there is a good chance your jack is fucked up. Maybe some dirts inside or faulty solder/contact.
You can try following the factory reset procedure, but I don't think it's going to work.

aneitzel, 2024-05-31:
Hi, i flashed around and also tried flash mayhem firmware new via DFU. Iam quite sure that this problem occured since flashing mayhem. 

What i not tested is deinstall the portapack from hackRF and flash it alone, but i assume that makes no difference at all?
Have no clue how to bring MIC TX back to live. 

:-(

gullradriel, 2024-06-27:
You can check and flash using the hacked cpld command line tool.
There is a section in the wiki.

gullradriel, 2024-06-27:
By the way you should read the Mic App wiki if not done, as it's talking about Mic settings.
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Microphone-Transceiver

gullradriel, 2024-10-06:
Closed due to inactivity.

closed 2024-10-06 by gullradriel