portapack-mayhem/mayhem-firmware #2159: GSM decoding feature?

Gbaby918, 2024-05-21:
### Description of the feature you're suggesting.

Recently, I have learned how to decode GSM traffic and was wondering if there was a way to add that kind of feature to the portapack? possibly a capture file or any suggestions?

phish27134, 2024-07-05:
learn what C or java?

eried, 2024-07-08:
What gsm decoding are you doing?

funder7, 2024-09-24:
@Gbaby918 if you give more details about what you mean by "decoding", and possibly how it's done (or the software used), then I think it will be easier to give an answer. I guess it all depends on available computing power.

gullradriel, 2024-11-11:
For reliable GSM signal analysis, a powerful processor (e.g., ARM Cortex A-series or an x86 processor) with at least 512 MB to 1 GB of memory would be more practical.

So, capture raw for future analysis on your computer, yes, doing it live or async on the portapack, nope no.

closed 2024-11-11 by gullradriel