portapack-mayhem/mayhem-firmware #2160: Themed GUI

htotoo, 2024-05-22:
### Description of the feature you're suggesting.

On Discord there was an idea, that tha whole GUI could be themed. Like all widgets, in all apps.

The idea:
Get a Style class, that has some pre defined names, like ForegroundColorNormal  for text color on widgets, that is not selected.
And define some more (the needed minimal that can handle all cases).

When the user changes theme this Style's properties would get overwritten, and the whole UI could change.

With this not only the Menu color could be changed, but the whole UI.

I prefer to ship some pre defined themes with the FW, and not all props could be changed by user. (no pmem error when everything would be white for example like text color and button background, ..). Just store the index of the selected theme.

There are some options, please vote:
- Don't do it.
- Do it, and remove menu color too, since the theme could set that too.
- Do it, but keep the menu color as it is now, and the teme would only change everything else.

### Anything else?

Submit any other suggestions with this idea in comment.

zxkmm, 2024-05-22:
I prefer 1 or 2.
I think the third option doesn’t make sense, because no one will set different color scheme for homepage and other elements, because it will looks extremely ugly.

zxkmm, 2024-05-22:
Also please vote on this post, if we should remove the current old-school windows XP style embossed button. And use flat style instead.

Yes, remove. (thumbs down emoji)
No, keep. (thumbs up emoji)

Radiomix2000, 2024-05-22:
The main features should be useful and practical. We don't have infinite amount of memory.

htotoo, 2024-05-23:
It should take max 20kb I think.

ToxxikHalo, 2024-05-23:
If this is an idea that's within the realm of feasible given the memory limitations and has enough interest, id kindly ask, because im closer to noob than not with the FW and coding  if someone has the time to put up an example pic of what gui "look" is possible... if that's an asinine thing to ask my sincere apologies in advance,lol. 
I'm just not knowledgeable enough of the HW/FW limits to know what's a reasonable expectation...

htotoo, 2024-05-25:
Some examples:

[image]
[image]
[image]
[image]

gullradriel, 2024-05-25:
Nice

htotoo, 2024-05-25:
Also figured out, how to keep the theme and the menu color problem in hands.
There will be a checkbox if the user wanna set the menu color too to the theme or not :)

htotoo, 2024-05-25:
Each new theme uses ~680 byte of flash size!
The whole theme system uses around 5500 byte.

closed 2024-05-27 by gullradriel