portapack-mayhem/mayhem-firmware #2163: About AGM chipe program

cn0xroot, 2024-05-26:
### Description of the feature you're suggesting.

I purchased a PortaPack H2. During use, the AGM chip was damaged, so I replaced it with a brand-new AGM chip myself. However, the screen does not display anything. Does this chip need to be programmed? How should I proceed? Thanks

cn0xroot, 2024-05-31:
agm ag256sl100 
http://www.agmsemi.com/program_details_5449993.html

zxkmm, 2024-11-13:
I had never play intel or intel-compatible CPLD.

But for common procedure, you write or generate the HDL code in IDEs like vivado (in this case, use the intel one instead, but I forgot the name)

Then flash it with programmer (usually each brand has its own programmer), by connecting the port (usually jtag)

So if you don't want to code yourself, what you should do is asking whoever sold the portapack to you, for the code or a programmed chip. Because we didn't write that so we don't have them.

WXLN, 2024-11-16:
我就说嘛，看这名称这么眼熟，原来是雪碧，这个问题貌似解决了，因为刚公布的h4里面，他把ag256sl100固件给分享出来了。嘿嘿，又一位刚老邵的同道中人