portapack-mayhem/mayhem-firmware #2169: Permission denied

myval, 2024-05-28:
### What happened?

Hi,
I have tried this  How to compile faster with WSL 2
https://github.com/portapack-mayhem/mayhem-firmware/wiki/How-to-compile-faster-with-WSL-2
but I am getting this error and not sure how to fix it, can someone help please?

[100%] Built target application.elf
[100%] Generating application.bin
copy from `application.elf' [elf32-littlearm] to `application.bin' [binary]
/bin/sh: 1: ../../../firmware/tools/export_external_apps.py: Permission denied
firmware/application/CMakeFiles/application.dir/build.make:60: recipe for target 'firmware/application/application.bin' failed
make[2]: *** [firmware/application/application.bin] Error 126
make[2]: *** Deleting file 'firmware/application/application.bin'
CMakeFiles/Makefile2:920: recipe for target 'firmware/application/CMakeFiles/application.dir/all' failed
make[1]: *** [firmware/application/CMakeFiles/application.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

### Tests

- [ ] I managed to update/boot previously

gullradriel, 2024-06-13:
Maybe ask on discord too

eried, 2024-07-24:
@myval did you manage?

closed 2024-08-16 by eried