portapack-mayhem/mayhem-firmware #2171: Meshtastic

PaulOckenden, 2024-05-31:
### Description of the feature you're suggesting.

I see that someone has already worked out how to do LoRa/Meshtastic using an SDR. See https://gitlab.com/crankylinuxuser/meshtastic_sdr - it’ll work with a HackRF. 

What are the chances of porting this to a Mayhem based Portapack as an app on the SD card? That would be so cool…

eried, 2024-06-07:
I think the chances are based on how popular is the idea to get enough people poking on it, but it could start with [one](https://github.com/portapack-mayhem/mayhem-firmware/wiki/How-to-collaborate#coding-new-stuff-or-fixing-bugs) :)

neednotapply, 2024-06-14:
I will note that in the article they created that nice Gnu Radio script and then said;
> We only need to throw CPU at the problem.

Easy for a PC but the Portapack does not have a lot of spare CPU.
That being said, we have a pretty amazing team 👍

archef2000, 2024-10-26:
The CPU part is mostly because it can decode all 8 LoRa presets at the same time and even `up to all of 900MHz ISM spectrum for all LoRa channels` which then needs more CPU to decode everything at the same time and `Real hardware can only decode the preset in which its set to.` so it would be less of a problem.

closed 2024-12-04 by gullradriel