portapack-mayhem/mayhem-firmware #2172: disk full

linebullet, 2024-06-01:
### Describe the bug.

Capture folder blank, but disk full.
v 2.0.0
[image]
[image]
[image]
Continuously capture half an hour.

### Reproduction

Press button capture.

### Expected behavior

1. Capture signal.
2.  Replay signal.

htotoo, 2024-06-01:
As your windows shows, the sd is full. Maybe really full, maybe just needs a formatting. But the "copy to sd" folder is not needed in that format.(You need to copy it's content to the root of sd), so you can delete that.

eried, 2024-06-04:
or corrupted sd card?

linebullet, 2024-06-04:
> or corrupted sd card?
Don't know.
How can I write the file structure to the new card?

eried, 2024-06-05:
as @htotoo format it and copy the "copy to sd" folder. But maybe try another sd card

linebullet, 2024-06-06:
> format it and copy the "copy to sd" folder. But maybe try another sd card

I copy filesystem to another sd card but fileman, write - "path: empty sd card!"
Capture -> rec, write "no filesystem".

htotoo, 2024-06-06:
You need to format it to fat32, check if it is that fs.

linebullet, 2024-06-06:
> You need to format it to fat32, check if it is that fs.

Not can fat32, microsd 128gb.
[image]

htotoo, 2024-06-07:
YEah, fat32 can't do that big partition. try to create a smaller partition (32gb for example) on the sd's first partition, format that to fat32. if you wanna use the rest of the sd, you can create a second exfat to there. but portapack won't see the second.

eried, 2024-06-07:
I am going to close this issue because it is not really related with the fw, but feel free to reopen if something related comes in

closed 2024-06-07 by eried