portapack-mayhem/mayhem-firmware #2176: My portapack enjoys annoying me.

Botox200, 2024-06-04:
### Describe the bug.

For some time now, the Portapack has been behaving completely randomly. Either the screen freezes immediately after starting up, or input via the jog dial is delayed for up to ten seconds. And sometimes (rarely the device works perfectly.

I have now made every possible attempt to narrow down the error:
1. Applied the entire troubleshooting area (Crash Guru Meditation).
2. Reformatted the card.
3. Bought a new card.
4. Changed all settings.

### Reproduction

1. Switch on
2. Jogdial input hangs or input latency up to 10 seconds

### Expected behavior

1. Switch on
2. Jogdial works

### Environment/versions

I have currently installed version 2.0.0, a 32 GB SanDisk Ultra, the original Portapack hardware and a Hackrf-One China clone. I currently don't have a coin battery in place.

### Anything else?

The Portapack worked perfectly last year. I imagine that this behavior of the Portapack only occurred after the newer versions 1.9 and later were installed (but of course that could be a coincidence).

zxkmm, 2024-06-05:
hello, 
you mentioned that "The Portapack worked perfectly last year"
which i assume you meant that same hardware (portapack+hackrf) works good last year?
So can you try downgrade fw back to 1.6 and see if it works normally?

and also: can you check if the device freezes without sdcard?

btw i'll edit the title for better read.

Botox200, 2024-06-05:
The error also occurs when the SD card is not inserted. Even a downgrade didn't make a difference. I did some tests with other Portapacks yesterday. The error didn't occur here. So it's clearly a hardware issue. Thank you.

closed 2024-06-05 by Botox200

zxkmm, 2024-06-05:
you are welcome. 
however it's still weird that your device sounds like eventually broken?