portapack-mayhem/mayhem-firmware #2178: Automatic start of hackrf when conneted to a host PC. It will help to remotely work on hackrf

umeshbisht77, 2024-06-11:
### Description of the feature you're suggesting.

Automatic start of hackrf when conneted to a host PC. It will help to remotely work on hackrf

zxkmm, 2024-06-11:
it could be done on PC sides, that when detect a serial device with specific name, send serial command to enter hackrf mode.
mostly i think it would be painful to have it as a function in Portapack side cuz p.mem is limited.
also detect the device is a charger or a computer would be painful as well, that if you enable this, when you charge, it will stay in hackrf mode

htotoo, 2024-06-13:
I would also prefer the serial method, since once you set it to HackRF mode, it'll start in it, and can't set it back. (unless you remove the ini file from SD, but that is not the nicest method, so users would be confused).

closed 2024-06-13 by zxkmm