portapack-mayhem/mayhem-firmware #2180: Microphone Transceiver: TX filter bandwidth is 6kHz. Should be 3kHz maximum

ly2ss, 2024-06-13:
### Describe the bug.

HackRF One r4. Portapack H2+  fw: Mayhem n_240608.
USB/LSB transmit filter bandwidth is 6 kHz. Should be 3kHz maximum. Combined with low cut off frequency 300 Hz would fit as designator 2K70J3E as per 
Recommendation ITU-R M.1732-3 (02/2023)
"Characteristics of systems operating in the amateur and amateur-satellite services for use in sharing studies"

More details:
http://www.rfcec.com/RFCEC/Section-3%20-%20Fundamentals%20of%20RF%20Communication-Electronics/33%20-%20TRANSCEIVER/Transceiver%20-%20ARS%20Approximate%20BW%20of%20a%20SSBSC-AM%20RTPY%20Signal%20Emission%20Designator%202K70J3E.pdf

Zilvis (LY2SS)

### Reproduction

The audio sweep test.
https://youtu.be/o7e-1pjucLg

### Expected behavior

Should be 3kHz maximum. Combined with low cut off frequency 300 Hz would fit as designator 2K70J3E as per  Recommendation ITU-R M.1732-3 (02/2023).

### Environment/versions

HackRF One r4. Portapack H2+  fw: Mayhem n_240608

Brumi-2021, 2024-06-18:
Hi @ly2ss,  you are right, the max  SSB bandwidth should be 3khz . Those switching filters are working well in RX side . I will have a look in TX side if we can reduce easily.  
73 cheers

Brumi-2021, 2024-07-06:
We can close now this issue with above merged commit . 
Later I will also update the Mic Transceiver wiki with that new feature .

closed 2024-07-06 by Brumi-2021