portapack-mayhem/mayhem-firmware #2193: Digital TV receiver/ Transmitter

Howdyowen, 2024-07-10:
### Description of the feature you're suggesting.

Some Way to watch digital t v on portapack

zxkmm, 2024-07-11:
Receiver had altered been implemented. Tx would be a nice feature request

eried, 2024-07-12:
Maybe he refers to ATSC? I do not think decoding would be easy/possible

htotoo, 2024-07-13:
Maybe DVB-T / DVB-S. Would be nice, but seems impossible on PP.

ACross99, 2024-07-20:
In germany dvb is h.265 (hvec) coded…in other countries might be h.264
First thought was, the mcu can ‘t handle this 
After a look at the specs…I‘m not sure anymore
But I think that there is a lot more ram needed to realize an encoding (respectively coding for tx)

just my 2ct…

gullradriel, 2025-02-04:
This would be impossible with the hardware ressources at disposition.

closed 2025-02-04 by gullradriel