portapack-mayhem/mayhem-firmware #2198: dmr+

elecpauto, 2024-07-17:
### Description of the feature you're suggesting.

Bonjour y a t il possibilité d ajouter une application pour décoder du dmr - dmr+ simplex directement dans le portapack H2

ACross99, 2024-07-22:
Good idea
Please check also for TETRA (Terrestrial Trunked Radio)

Ryzard-99, 2025-01-06:
Hello! Is it possible to integrate DMR (and possibly TETRA) decoding directly into the PortaPack Mayhem firmware? I'd be happy to help test prototypes, provide feedback, and collaborate to see if we can make it happen!

pauldonahue, 2025-01-09:
Another vote for allowing this!

elecpauto, 2025-01-09:
Avec le logiciel sdrangel et le module dsd cela fonctionne mais quelqu un
pourrait il intégrer ce module dsd directement dans les applications du
portapack

f1ghy, 2025-01-09:
Hi, It could be nice, of course. It is not the first time this feature is asked. I am afraid it is a too hard work for the small memory of our Portapack. Maybe now there is external app ?

gullradriel, 2025-01-09:
Let me tell you that real-time audio decoding is very unlikely due to the CPU and RAM constraints.
Using the Portapack in hackRF mode along with a raspberry pi is the most portable thing you can get.

elecpauto, 2025-01-10:
Merci

gullradriel, 2025-02-04:
Decoding DRD (Differential Radio Data) and DSP+ (Digital Signal Processing Plus) directly on the HackRF Portapack is likely to be very challenging due to hardware limitations. Here's why:

# 1. Hardware Constraints of HackRF Portapack
- Limited Processing Power: The HackRF One (even with the Portapack Mayhem firmware) relies on an ARM Cortex-M4 microcontroller (STM32F4). This MCU is great for handling control logic but lacks the raw DSP power needed for real-time signal decoding beyond simple modulations.
- No FPGA or DSP Accelerator: Unlike SDRs like the USRP or LimeSDR, the HackRF doesn't have an FPGA or dedicated DSP processor. It relies purely on software processing.
- Memory Limitations: Your firmware already faces space constraints (4668 bytes left in flash). Complex demodulation might require additional libraries or look-up tables that may not fit.

# 2. Feasibility of DRD and DSP+ Decoding
## DRD: If it is similar to other differential encoding schemes, decoding could require:

- Synchronization to a reference signal.
- Phase-difference calculation over time.
- Potential error correction (which might need extra processing).

If DRD is purely differential and relatively simple, it might be possible with heavy optimization.

## DSP+: If this refers to an enhanced DSP modulation/demodulation scheme, it might involve:

- High-order QAM or PSK demodulation.
- Advanced equalization and filtering.
- High-speed FFT or convolution operations.

These typically require floating-point operations, which the STM32F4 lacks in hardware (though it supports software floating-point).

# 3. Potential Workarounds
- Preprocessing on a Host PC: Stream raw IQ data from the HackRF to a PC and handle the decoding there.
- External Co-processor: If the decoding is essential in an embedded setup, an ESP32-S3 or an FPGA could assist with DSP tasks.
- Highly Optimized C/C++ Implementation: Stripping down the DSP processing to integer math and LUT-based calculations might make a basic implementation feasible.

# Conclusion
Unless DRD and DSP+ are very lightweight modulation schemes, real-time decoding directly on the HackRF Portapack is unlikely to work efficiently. Offloading the heavy lifting to a more capable processor is probably the best approach.

closed 2025-02-04 by gullradriel

elecpauto, 2025-02-04:
> <https://github.com/portapack-mayhem/mayhem-firmware/issues/2198> as not
> <portapack-mayhem/mayhem-firmware/issue/2198/issue_event/16175365170@
> github.com>

Merci pour cette réponse détaillée pour l instant je l utilise en connexion
avec un pc et le logiciel sdrangel cela fonctionne bien mais reste
difficilement mobile

KvaKKK, 2025-06-17:
Hi all. May be some code from https://github.com/sannysanoff/SDRPlusPlusBrown/releases/tag/rolling can help with DMR decoding. 
Also ask if it possible to and DMR listening feature to FW

piraslettis-tech, 2026-04-24:
Sono molto interessato all'argomento