portapack-mayhem/mayhem-firmware #2200: Pacman? (No bug)

sizzlik, 2024-07-18:
### Describe the bug.

Hi,
what happened to the pacman app? Seems its not included anymore..but the greyed out pacman menu icon still remains. Any chance to bring pacman back..or how do i remove dead menu entrys?

### Reproduction

Check app folder, no pacman file..grey menu entry

### Expected behavior

Have the app or have the menu entry removed

### Environment/versions

newest nightlys

gullradriel, 2024-07-18:
Looks like you just updated the firmware without updating Sdcard contents.
If you use the mayhem hub webpage to update, it should update the apps.
If you just flash the bin, the external apps will stop working.
Note that Pacman have been promoted to a new type of external app, you may also have a bad old app file on your SD card.

sizzlik, 2024-07-18:
> Looks like you just updated the firmware without updating Sdcard contents. If you use the mayhem hub webpage to update, it should update the apps. If you just flash the bin, the external apps will [...]

Dunno what went wrong then..im doing the same every time..replace all files with every new nightly (without world map) and then flash with the internal flashing tool. If no big changes are made i just download the tar that contains apps & firmware. Did that aswell today. I was abroad last few weeks and just recieved a brand new original hRF r10 plus a fancy portapack that got all your names printed on the pcb (you know about this? Are online aliases copyrighted?) So i just took the good sd from the other PP, copied the new nightly sd content over the old one and flashed. Worked as expected, just pacman was missing. What is the mayhem hub webpage? Never used it before..just internal or hackrf_spiflash.

sizzlik, 2024-07-18:
Edit..i just noticed..pacman is still there..but i got a double entry
[image]

How do i get rid of that entry?

gullradriel, 2024-07-18:
I recommend deleting all pacman files on your SD card, then update SD card content with the zip associated to the bin you're using.
Or just delete the apps and then update using Mayhem Hub:
https://hackrf.app/
As it's updating both bin and Sdcard apps.

ACross99, 2024-07-18:
I wrote this weeks ago…

the tar package includes the pacman, the zip unfortunately not

just pick it from the tar container, copy to where it should be

closed 2024-07-18 by htotoo