portapack-mayhem/mayhem-firmware #2215: Been trying for months and just cant get it

DrewMeager, 2024-08-09:
### What happened?

I have been trying to update the firmware on my Portapack H2+ for months. I have tried everything I have found to do including but not limited to separating the boards, putting it in HackRF mode, putting it in DFU mode, etc. I was able to get the files for the sd card onto the SD card, but not on the unit itself. It came with version 1.9.1 already on it, I am just trying to flash the latest stable version (2.0.1). I am at the point where if I could just send it to someone I would as it has been a headache the entire time.

### Your hardware

H2+R4

### Tests

- [ ] I managed to update/boot previously

htotoo, 2024-08-09:
What were the results of you tries?
Explain a bit more please.

DrewMeager, 2024-08-09:
the closest I got was my terminal saying "dfu-util: Could not open file hackrf_one_usb.dfu for reading: No such file or directory
Drews-MBP:~ drewmeager$ dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
dfu-util 0.11". 

my computer (Macbook Pro running 10.15.7 Catalina) doesn't seen to recognize that the HackRF is even plugged in no matter what I do. I understand that this is really annoying to do, but unfortunately it may be a case where I need step-by-step direction.

htotoo, 2024-08-10:
Use linux or windows if you wanna update from a computer. More reliable. Put in an sd card to pp, and via hackrf.app webpage.

If you can't access a pc, then insert the sd card to the mac, and download the tar file from github releases page, but NOT with safari! (That extracts the files, and we don't want that). Copy the tar file to sd/firmware folder. Eject sd, and insert into pp.
Then use pp's own flasher utility...just select the downloaded file.

This should work.

DrewMeager, 2024-08-10:
I was able to get it to connect to the hackrf.app webpage after going to a windows machine (thank you), but it is saying that the firmware version is too old to manage it through that. if I utilize the second option you gave me, would I be able to use the webpage for future updates?

 [image]

htotoo, 2024-08-10:
Ohh, then you can't update it with the ppfw.tar file, so first you need to do it with the old way.
Copy the .bin file from the releases to the sd/FIRMWARE folder, and use the pp's own flash utility.
Then you should copy all the copy to sd.zip's content to the root of the sd card.

After it you'll be able to use the website to update.

DrewMeager, 2024-08-10:
I was unable to find the .bin file you are talking about. just for fun I decided to look in the files on the SD card. there is a firmware folder on the root of the SD card (which has nothing in it), but also one in the "mayhem_v2.0.1_COPY_TO_SDCARD" folder (which has "portapack-mayhem_v2.0.1.bin")

[image]
[image]
[image]

htotoo, 2024-08-10:
Use that bin. Or the one from this, this is newest: https://github.com/portapack-mayhem/mayhem-firmware/releases/download/nightly-tag-2024-08-10/mayhem_nightly_n_240810_FIRMWARE.zip

DrewMeager, 2024-08-10:
That worked nicely, thank you so much for your help!

DrewMeager, 2024-08-10:
Now I need to find the web page where it tells me what all the applications do.

htotoo, 2024-08-10:
Read the Wiki here on this github page. That helps a lot!

closed 2024-08-10 by htotoo

DrewMeager, 2024-08-10:
You have been absolutely amazing! I will have to do that, thank you so much!