portapack-mayhem/mayhem-firmware #2219: HackRF One w/ software version c8774e82 not taking update

6051H, 2024-08-10:
### What happened?

New out of box HackRF One with Portapack H2 (aluminum frame) with software version c8774e82 is not taking updated software version v2.0.1. Can't get any computer to recognize device. DFU boot and dfu_hackrf_one.bat does nothing on a windows machine. Any help, sigh, step by step, or specific video assistance greatly appreciated.

### Your hardware

HackRF One with Portapack H2

### Tests

- [ ] I managed to update/boot previously

htotoo, 2024-08-10:
Hello.
Seems to be duplicate of this (kind of) https://github.com/portapack-mayhem/mayhem-firmware/issues/2215

Tldr:
Try to update it via copy the bin file to sd card and update with the pp's own flasher utility. 
Later you'll be able to update it with the hackrf.app site (after you updated to the latest version)

6051H, 2024-08-10:
> Hello. Seems to be duplicate of this (kind of) #2215
> 
> Tldr: Try to update it via copy the bin file to sd card and update with the pp's own flasher utility. Later you'll be able to update it with the hackrf.app site (after you updated to the latest version)

In the Utilities menu my PP only has the following 6 submenus: 
1) Freq. manager
2) File manager
3) Signal gen
4) WAV viewer
5) Antenna length
6) Wipe SD Card

No flasher utility.  Any thoughts other than the f*ck-it bucket?

htotoo, 2024-08-10:
Then try the usb mode: https://youtu.be/qM7Q9IidK2o?t=384&si=59HuSpn5L6Bapsgg

If this don't find the hackrf, then you'll need to install the driver for it and try again.
In that video there are other modes (like chapter 5 the dfu mode), that you could try.

ACross99, 2024-08-11:
Do you have a WORKING sdcard inserted?

keithgold, 2024-08-16:
I had similar issue.  Turned out to be a bad cable!!  I tried 3 of them and the last one passed data properly.  The others were only good for power!  If it’s not the SD Card, CHECK THE USB CABLE!!

eried, 2024-08-19:
Thanks for sharing @keithgold, It is a good hint for @6051H to test.

Was the cable included with your device?

gullradriel, 2024-11-11:
Closed due to inactivity / lack of answers.