portapack-mayhem/mayhem-firmware #2220: Plenty of bass in TX

itaniepitas, 2024-08-13:
### Description of the feature you're suggesting.

Is there any way to fix this problem ? Lots of bass without any treble or hi audio frequencies on TX with any settings. It is almost an unintelligible transmission due to bad muffled audio. Acoustic frequencies not existed above 400 Hz. I believe that a more linearity settings is needed in the voice processing. Unless the problem is with my own device.

eried, 2024-08-14:
That sounds a problem with the microphone, is it an integrated mic? or external?

itaniepitas, 2024-08-14:
Both Erwin. First I trying to change the electret element in portapack H2M with different types of elements. No success.  After that i change the bridge 0 Ohm resistor to make the speaker - mic jack hot. Different external mic also no success. Checking the circuit for some modification to mic input with a lower value capacitor in WM8731 but all ready use a small capacitance of 1nf. No further ideas.   
Chrys

eried, 2024-08-14:
do external mics sound the same? can you show an example of both?

itaniepitas, 2024-08-14:
I hope to you catch the difference in audio waterfall in video. The first transmission of  French guy how look the audio and the second my voice. The bandwidth is  3KHz in SSB. I use an external microphone. As I describe before, no difference or the internal is worse versus to external. Needs middle and high frequencies to fill the bandwidth with audio. 

https://github.com/user-attachments/assets/9125ef5a-23ed-47c6-9e1e-f63c1b2546ec

eried, 2024-08-16:
Oh strange, clearly does not sound OK. I hope someone here can pitch in with a similar experience, it has not been reported before

itaniepitas, 2024-08-16:
Hi
First of all let's make something clear. Before the Nightly Release 240707 - 240709 all previous firmware with bandwidth over 3KHz was unacceptable. Period. Before this fix, the audio was a nightmare in a huge bandwidth with distortion spurious mirror  images etc .  With the current settings it works perfectly in ssb emission limits. Everything is fine except the microphone is unable to produce during speech all those required frequencies to fill this bandwidth. Unless there is a problem with portapack hardware, because don't use all this components that the manufacturer shows in the schematic drawing for the correct input function of the microphone. Something like that. 
 
[image]
I will investigate further this section
Chrys

Brumi-2021, 2024-08-16:
Ok @itaniepitas , now is clear. 
Thanks to the ham colleague  ly2ss, we managed to limit and test  the Audio SSB TX to 3Khz . 

Initially I allowed 2k,3k,4k .(n_240707 )
 But clearly 4k was doing some obvious artifacts , and we both decided to  delete 4k option in second commit  (n_240709). 
Further investigation showed me  that also 2khz option is producing some folding artifacts . (But just when audio in BW audio in is beyond 4khz , and so far we did not removed that option) 
 I could see that we can easily have perfect BW cut off without any artifacts in BW   3khz and 6khz . (But 6khz is too much over ham SSB spec) .

I checked with my Mic (and old apple iPhone headphone -mic set) latest commit  and with my Tecsun PL660 the transmitted audio TX SSB audio sound. Quality was quite good . 

Waiting for your further test.
Cheers

itaniepitas, 2024-08-17:
Something else. Possibly due to the low signal processing ability, only 8bit sampling rate, cannot properly digitize the entire required audio spectrum on 3KHz bandwidth. What I could suggest would be something similar to what is used in SdrAngel software. In other words, the choice to select a preset or by slide bar a low cut-off frequencies like 100Hz - 200Hz -300Hz etc. which have the effect of releasing processing power from these frequencies witch after all, do not offer anything essential in telecommunication. Also a Hi cut-off would be nice to expand the range in KHz without the presets of 2 or 3 KHz bandwidth. With this method the user has the ability to adjust the transmission by the different mic - voice - etc.  Just my thoughts. My research in hardware continued.  
Chrys

gullradriel, 2024-12-20:
Closed due to inactivity, and not planned.

closed 2024-12-20 by gullradriel