portapack-mayhem/mayhem-firmware #2226: Portapack H2 19kHz stereo pilot at Headset

IvanRadolo, 2024-08-18:
### Description of the feature you're suggesting.

Now high frequenies for some reasons are blocked so there is not possible to see the 19khz and other freq that can portapack give out to the headset output. Maybe also would be possible to output all freq and also 57kHz RDS modulation.

### Anything else?

Is possible to add a stereo decoder so stereo can be applied to the headset output?

zxkmm, 2024-08-19:
> Now high frequenies for some reasons are blocked

Could you please explain this more clearly? Define "blocked out", and which specific frequencies?

> Maybe also would be possible to output all freq and also 57kHz RDS modulation.

This freq is too low and out of hackrf freq range. both 57kHz and 19khz.

IvanRadolo, 2024-08-19:
When I record HackRF headset output with PC audio editing software I don't see stereo pilot signal at 19kHz on the recording, when recording standard radio it pass all the frequencies.

No, its not low freq because it do not need to receieve it, its on the listening frequiency rang on the headset output, it's deviation frequency and single channel audio is from 0 to 16kHz then starts pilot at 19 khz and then codes the difference between left and right channel after the 19 kHz pilot code. It's the MPX coding method.

zxkmm, 2024-08-19:
I got you.  
Sorry about my misread, I don’t quite familiar with this field and also I don’t quite know what’s MPX coding.
I hope @Brumi-2021 have more information about your request.

Brumi-2021, 2024-08-19:
Hi @IvanRadolo , 

Let me try to clarify it , adding  two comments about it : 
1-)  when we are tuning any commercial FM with Audio RX App  , we can see on top of the waterfall  a part of the the FFT MPX  spectrum . In that part FFT is runing with fs sample rate of 96khz , that is why we can see a max. FFT of 48khz (fs / 2 )  . 
Therefore we can see  the 19khz stereo carrier tone , and also the DSB (L-R) on 38khz (that should work in current FW)  ,  but nowadays we can not see the RDS 57Khz .   I also would be interested to see that RDS 57Khz tone .  (Technically it should be possible if we make work that  FFT Spectrum piece of code (with some changes)  at example double SR.  96khz x 2 . Someday I may try it ...) 

2-) But in any case the recorded audio base band are limited to 15khz , as any FM  receiver.  (Mayhem is not recording the MPX spectrum , just the audio base band .  (but when you Capture that RF part for Replay , you are recoding all, including  MPX ) 

Regarding the  FM  stereo decoder, I know how to do it using  Hackrf + GNU Radio , but regarding  direct C++ encoding in Mayhem , ...,  I do not see it easy ... and moreover , now Mono FM decoding + Spectrum is already in the limit of CPU capacbility . Maybe removing  the FFT spectrum we may have enough MIPS to do it ... I can help, but I can not implement it alone . We would  need more skilled sw engineers to implement it .    

I hope , that you followed me , 
Cheers,

IvanRadolo, 2024-08-19:
Hi @Brumi-2021,

I believe this is an unusual issue, not a typical bug.

As I mentioned, there is no stereo pilot signal on the analog headset output of the PortaPack. I recorded the output with my laptop, and it’s just pure mono with noise around 16 kHz—no 19 kHz pilot signal visible on the oscilloscope. However, when I record the output from my small mono battery radio, I can clearly see the 19 kHz pilot. How is it possible that the 19 kHz pilot is missing on the HackRF PortaPack headset output?

By the way, it would be great to have an option to disable the spectrum and add stereo decoding.

I have an off-topic question: since the HackRF currently doesn’t support stereo output, is there any other SDR device that can decode MPX to a stereo signal standalone? I have an audio-encoded MPX signal at 2 GHz.

Thank you,
Ivan

Brumi-2021, 2024-08-21:
Hi @IvanRadolo , 
as I told you in my previous comment.
Our fw is correct. It is design concept .(and I fully agree with tut).

We are delivering to the headphone jack only the signal part that is audible , therefore we intentionally are filtering and  cut the pilot tone 19khz and DSB (L-R) of 38khz . (And same concept is applied into the WFM mono REC WAV files)

BTW , Why do you need that MPX signal in the headset output ?  Do you need it also in the recorded WFM REC WAV signal ? For which purpose ?  We can already decode FM stereo signal with any dongle SDR or HackRF and external SDR program and PC . To me the only benefit would be to have standalone Portapack decoding or not .

Depending on your answer   , let’s consider potential enhancement feature , but personally I do not see so much benefit. (to me, there is not big deal , just send the MPX signal to the headset output jack ). 

Regarding your recorded stereo audio signal from 2Ghz , in which format do you have it ? If it is in IQ raw signal , for sure that any SDR PC program can decode it .

cheers

Brumi-2021, 2024-08-21:
Hi @IvanRadolo ,
Ok , are you available in discord ?
I can easily check the LPF 15khz bypass or modify it to handle the 19khz , but that is not enough... 

 I need to check the SR that we are using in WFM  for the headset  . But as far as I remeber ,  I think that we are running the audio codec (WM8731  or AK4951K) at 48khz in WFM , (24khz in NFM and 12khz in AM)  , in that case, now  the max audio headset output bandwith would be 24khz.  If so , just bypassing the 15Khz LPF ,  we could send  the 19khz tone , but not the full MPX DSB @38khz .

  I think we will need to increase the audio codec SR to 96khz to handle it , and that needs to be carefully check . 

Cheers

IvanRadolo, 2024-08-21:
Hello @Brumi-2021 ,

Add me as a friend on Discord! Invite expires in 1 week: https://discord.gg/9vVe49kD

Strange that is cut at 15kHz as usually more than 17kHz is not audible and also most radio stations goes up to 15,3 kHz - 15,7 kHz, with new audio processors it's possible to receive 17 kHz with stereo.

MPX signal also contains RDS data at 57kHz for that SR must be 192 kHz, I don't know how much portpack card analog output can support.

Thank You,
Ivan

Brumi-2021, 2024-08-21:
Hi @IvanRadolo 
I think we can close that request 

As we have discussed by direct chat, to handle composite Multiplex signal (MPX out, through headset jack , we will need to run SR audio codec to 192khz , but currently WM8731 (SR 96KHZ max) , AK4951K (SR 48KHZ max) , therefore , due to hw device limitations, we can not handle that enhancement issue request . 

MPX = Audio baseband + 19khz tone + DSB 38Khz suppressed carrier (23k to 53khz L-R) + 57Khz RDS BPSK signal . 

Sorry about that !

closed 2024-08-21 by Brumi-2021