portapack-mayhem/mayhem-firmware #2227: The suggested device in readme is no good!

fehmi, 2024-08-19:
### What happened?

I ordered it from Aliexpress. The decime came with this following issues.

- Not booting if it's not fully charged. This limits user only for usb connected use.
- Saving settings is not working. It resets on every boot. Changing cell battery didn't help.
- Tried everything but headphone jack is not working. 

Do you know any solutions for them? 

That was a horrible experience for me to start portapack. Maybe you should change the device in readme with H2M (Mayhem edition) if it's better. I ordered one and waiting for it. I hope it will work better. 

### Your hardware

Portapack H2+

### Tests

- [X] I managed to update/boot previously

eried, 2024-08-19:
Oh, lame. The links from the readme were the ones where I ordered 2 units I have, thats the reason those are there (the design of the case, and all initial tests were on those, many years ago). We are in the process of preparing a version with a manufacturer to have a "more" official one, but this might take time.

Contact aliexpress to complain, the seller should help you.

eried, 2024-08-19:
I am gonna close this issue, since is not about the firmware, but it can be reopened if needed.