portapack-mayhem/mayhem-firmware #2228: HackRF One r9: "Hardware does not appear to have been manufactured by Great Scott Gadgets"

Navneethmenon, 2024-08-21:
### What happened?

"Hardware does not appear to have been manufactured by **Great Scott Gadgets**" Message after Flashing **Mayhem Firmware**
After successfully flashing; 

Already referred to [#1442](https://github.com/greatscottgadgets/hackrf/issues/1422) #

```
hackrf_spiflash -w portapack-h1_h2-mayhem.bin 
File size 1048576 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 1048576 bytes at 0x000000.
...
hackrf_info
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: 000000000000000042a068dc366dae07
Board ID Number: 4 (HackRF One)
Firmware Version: v2.0.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x00664766
Hardware Revision: r9
Hardware does not appear to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

```
Tried all these;
  mayhem_v2.0.1_FIRMWARE
greatscottgadgets/hackrf/release/v2024.02.1
greatscottgadgets/hackrf/release/v2023.01.1
greatscottgadgets/hackrf/release/v2022.09.1

Bought from [Robokits](https://robokits.co.in/)

I am facing some compatibility issues while trying to use the Mayhem firmware on my HackRF One with hardware revision r9. The hackrf_info command indicates that the hardware was not manufactured by Great Scott Gadgets, though it should still be supported with the given firmware. Some compatibility warnings are still thrown by the device after flashing.

### Your hardware

HackRF One 9/11/22 

### Tests

- [ ] I managed to update/boot previously

zxkmm, 2024-08-21:
Hello, so you meant that, when you flash pure hackrf fw from GSG (not here), you still have that incorrect warning?
If you still have that incorrect warning, I’m afraid that’s not from our side.

here’s something you need to check and reply:
1. Remove your portapack (it should not related but just in case), flash pure hackrf fw, and show us the output info here.
2. Submit an image o flash your hackrf and I’ll check if it’s original.
3. Double check if your hackrf is original.

Btw, it’s easy from here to show a fake info but I don’t think it make sense.

closed 2024-08-21 by zxkmm

zxkmm, 2024-08-21:
Based on your image, your Hackrf is a clone instead of original, so it's normal to see that warn.
[image]

zxkmm, 2024-08-21:
You are welcome. Clones works basically the same with original (except maybe the signal quality things). So you can use same firmware as original. If something not working, submit another issue ticket.