portapack-mayhem/mayhem-firmware #2238: Since 2.0.1 I can't save general settings

amelillo, 2024-09-07:
### Describe the bug.

Hello,
Since 2.0.1 up to the latest firmware (n_240907) I can't save general settings.
For example, Mayhem splash screen is shown even if I choose not to show it.
When I enter settings I can select it and save. If I exit and re-enter settings I can see the setting has been saved.
Then, if I power off and reboot, the splash is shown and the setting is on "enabled" again.
The same thing occurs if I select to use high speed mode on sd card: the settings appears to be saved and it's not when I reboot. Same for a theme: the PP defaults to the standard theme at reboot.
However, the sd card is writable as I can save, for instance, a new remote for my gate, and it appears on the list if I reboot.
Am I missing something?

### Reproduction

1 switch on
2 select a setting for user interface in the setting section and hit "Save"
3 power off
4 reboot
5 the setting hasn't been saved

### Expected behavior

User interface settings should be saved.

gullradriel, 2024-09-07:
That's a dead coin battery.

amelillo, 2024-09-07:
Oh, didn't think 'bout it. I'll try as soon as possible and see if it fixes the issue. Thank you!

gullradriel, 2024-09-07:
You can also use pmem to Sdcard, see the wiki

gullradriel, 2024-10-06:
Closed due to inactivity.
It's clearly due to a dying coin battery / use of pmem to sdcard option (like if you forget to get into Pmemory Mgmt and click 'save' to update the settings)

closed 2024-10-06 by gullradriel