portapack-mayhem/mayhem-firmware #2241: HackRF One is not recgonized by USB on any laptop or computer

AKProwl, 2024-09-07:
### What happened?

I can't detect my HackRF on any Computers to use a batch file or even a cmd prompt to flash the new updates

### Your hardware

Portapack HackRF One H2

### Tests

- [X] I managed to update/boot previously

htotoo, 2024-09-07:
What version you have?

Checked the wiki for all the update methods?

Are you using a proper usb cable? Have you installed drivers?

namdnas, 2024-09-20:
Have you placed the Portapack into HackRF mode so that the HackRF can be seen directly by the computer? Assuming you have you can check on a linux computer with `dmesg` or on Windows check with hardware manager. You should see when you plug it in on Windows a new `unrecognized` device until the driver kicks in. Also have you installed any drivers that could conflict with the HackRF connecting properly.

Trufax, 2024-09-22:
Had the same issue. Removing the Portapack, boot with USB plugged in HackRF with Reset and DFU pressed, release DFU and after that release Reset solved it. I could flash DFU, than Mayhem and reassembled the thing.

Neimar-tm, 2024-10-11:
check your usb rules with "lsusb"

reischle, 2024-10-21:
I have one or twice blown both the USB ESD protection Chip (U15) and the MCU (U23). Reason was a micro usb cable that - when jiggled - had the VCC/VBUS touch the data lines. Also killing the voltage regulator U21 in the process. Check for a telltale "crater" in U15.
If you are lucky, it might work with U15 removed. If the USB Interface of the MCU is blown, it can also be replaced with good soldering skills, but I failed on that - partly delaminating the PCB with my preheater. (The board still does "run", but the MCU does not see I/Q data.)
So when the Hackrf does not show up in lsusb (Linux) or Device Manager (Windows), consider a hardware fault around the usb connector.

gullradriel, 2024-11-11:
Closed due to inactivity / lack of answers.

closed 2024-11-11 by gullradriel