portapack-mayhem/mayhem-firmware #2242: TPMS not working

jonesea, 2024-09-07:
### Describe the bug.

No readings on 2.0.1 or nightly 2024-09-07, BUT DOES WORK on 1.7.4

Using 315 MHz with AMP:1 LNA:32 VGA:32

### Reproduction

Fresh install, defaults, Receive / TPMS Cars application.  Switch to 315 MHz, AMP:1

Either wait, or trigger TMPS sensor manually with external "learn" tool.

### Expected behavior

Screen should populate with TPMS sensors close by.

jonesea, 2024-09-16:
I tracked this down to the last working build was 2023-11-21.  I believe [1598](https://github.com/portapack-mayhem/mayhem-firmware/pull/1598) broke my receiving of TPMS.

From the notes:
> Set receiver to 314.95 MHz when configured for "315", since half of the TPMS sensors transmit on 314.9 and half on 315.0.

NotherNgineer, 2024-09-16:
The bandwidth of 1.75MHz should be sufficient to cover both 314.9 and 315 MHz, but I've taken your suggestion and modified TPMS RX app to allow the user to select between them in PR #2252.  A test version is attached if you could please try it.  To install, you just need to unzip the file so it's a tar file, copy it to the FIRMWARE folder on your SD card, and use the latest "Flash Firmware" app to install it.  Please let me know if this helps.

[portapack-mayhem_OCI.ppfw.tar.zip](https://github.com/user-attachments/files/17008412/portapack-mayhem_OCI.ppfw.tar.zip)

NotherNgineer, 2024-09-16:
This issue got closed automatically when I merged PR #2252.  If the test version doesn't fix this, then please re-open this issue ticket.  Thanks.

closed 2024-09-16 by NotherNgineer