portapack-mayhem/mayhem-firmware #2244: Rotary clockwise

Harrie1966, 2024-09-08:
### Description of the feature you're suggesting.

Hello, I have a problem with my receiver. The rotary is broken and the direction of rotation is reversed. By going to the left the frequency goes up and by going to the right the frequency goes down. What can I do to get this back to normal. Thanks in advance, hopefully someone can help me. Greetings Harrie

eried, 2024-09-09:
Was it reversed before or something happened?

NotherNgineer, 2024-09-16:
FYI, you can view the rotary encoder output using the Debug -> Buttons Test app.  When you turn the encoder slowly clockwise (using latest firmware) it should show "Dial: 1", or show "-1" if you turn it slowly counterclockwise.  Is that what you see?

You might also check the Settings -> Encoder Dial and see if the sensitivity is set to NORMAL, and rate multiplier is 1 (default).

I had to replace the rotary encoder on my first Portapack because I wore it out.  There's some info on the wiki pertaining to this.

gullradriel, 2024-11-11:
Closed due to inactivity / lack of answers.

closed 2024-11-11 by gullradriel