portapack-mayhem/mayhem-firmware #2250: Distorted POCSAG signal recorded by the Audio App

flux242, 2024-09-12:
### Describe the bug.

As nobody reacted on discord I'm repeating the same question here. Don't know if that's actually a bug or me not using device correctly or some hw problem

Hello, I have portapack for couple of days now and was trying to decode pocsag and it couldn't decode messages. Then I started the Audio app and recorded the POCSAG signal, then I recorded the same signal using gqrx with the same IF and VGA values. Portapack recording seems distorted and I don't understand what to do with it. Any hints? Was using NFM 11k portapack bw and 10k gqrx bw. Preamp was disable

Here are the pictures of the audio recorded https://imgur.com/a/BAVEF6K

### Reproduction

Start Audio app, set IF to 16, VGA to 24, tune to the local POCSAG frequency (466.075 here) and record the audio

### Expected behavior

Recrorded signal should look like the one recorded with gqrx

### Environment/versions

2.0.1

Radiomix2000, 2024-09-13:
I found similar issue with POCSAG. It looks like the transmitted signal paying speed is accelerated 2 times. Possibly interpolation/decimation/samplerate is somewhere broken.

flux242, 2024-09-14:
A AM OOK signal recorded with mayhem and gqrx. Obviously, a high pass filter is used instead of a low pass one somewhere in the demodulation chain. So NFM, AM demodulation is broken

[image]
[image]

flux242, 2024-11-09:
zt

flux242, 2024-11-09:
zt

closed 2024-11-09 by flux242