portapack-mayhem/mayhem-firmware #2257: EPIRB decoding

OlivierTE, 2024-09-21:
### Description of the feature you're suggesting.

Hello, in charge of decoding EPIRB beacons (COSPAS SARSAT), I carry out this task from traditional radio receivers and electronic decoders. The whole thing is heavy and tedious to transport. The portapack would be an easy to transport solution and probably very effective if decoding could be done directly in the software. Sorry if the feature already exists but I haven't found it.
Being part of the official research teams, I can have a test beacon if necessary.

htotoo, 2024-09-21:
Seems interesting!
https://www.sigidwiki.com/wiki/Distress_Radiobeacon_(Digital)
~406mhz, BPSK, 400bps

Sat info: https://www.itu.int/dms_pubrec/itu-r/rec/m/R-REC-M.633-1-199006-S!!PDF-E.pdf

htotoo, 2024-09-23:
If you got an IQ file (c16) capture with portapack, you can attach it here (more is better!). That way if someone wanna look into it, it will be easier.

OlivierTE, 2024-09-23:
Hello, I ordered it yesterday on Aliexpress following your recommendations.
As soon as it arrives, I will send you the file. Thank you very much. Have
a great day. Olivier

AVTMike, 2025-01-13:
Hey guys,

I'm an avionics engineer in Canada, and I had interest in a very similar function (in my case, perhaps obviously, capturing and decoding the FSK data transmitted by aviation beacons on 406MHz during a test pulse).  I was directed here from the Discord and I'd like to follow up with Olivier... Did you ever manage to capture a test pulse and upload it?  I'm very interested in helping develop this further, and I can provide a test 406MHz pulse from an aircraft ELT in the coming days if that's of help.

Thanks,

Mike

Brumi-2021, 2025-01-14:
Hi @OlivierTE and @AVTMike , 
On March 2024 @przemekmac requesred us  same feature , through discord. He sent me several sample files and related information . I studied it and got the some idea about it's implementation . But it's code  porting  was not easy for me .  Then I just postponed it  . Now i am busy with other features , But I may come back to it , but I will need coding help.

AVTMike, 2025-01-14:
Hi Brumi-2021,

Thank you for the update!  I appreciate the insight and hearing about past interest for this feature.

Unfortunately, I'm completely illiterate when it comes to most coding.  I'd love to offer my help but I fear it might not be most useful; it sounds like you already have some solid sampling and resources from another user.  Regardless, let me know if I can be of any other help and I'd be happy to do so.

Thanks again,

Mike

Brumi-2021, 2025-01-15:
Hi , are you also available through Hackrf Portapack Mayhem discord ? 
We can comment  directly about it . 
Cheers

loorisr, 2025-06-13:
That's great if you are in contact :) 
I agree with you, it may be difficult to use his code to decode the BSPK modulation on a Porta Pack. His code can be useful to decode the EPIRB information after the BPSK decoding.