portapack-mayhem/mayhem-firmware #2262: M0 Guru Meditation - “Hint: Out of Memory”

AKProwl, 2024-09-23:
### Describe the bug.

I added 550MB worth of 8 & 16-bit mono .wav filed between 24000 and 48000 Hz. Following the instructions on https://github.com/portapack-mayhem/mayhem-firmware/wiki/Soundboard

When I open soundboard, manually type in frequencies and other features it’s reads M0 Guru Meditation - “Hint: Out of Memory”.
“Press DFU for stack dump” doesn’t help or resolve the issue.

### Reproduction

1. Adding your own sound files using the instructions from https://github.com/portapack-mayhem/mayhem-firmware/wiki/Soundboard
2. Then turn on HackRF
3. Go to Transmit
4. Click on soundboard and M0 Guru Meditation - “Hint: Out of Memory”.
“Press DFU for stack dump” will display
[M0 Guru Meditation out of memory soundboard error.pdf](https://github.com/user-attachments/files/17094612/M0.Guru.Meditation.out.of.memory.soundboard.error.pdf)
[image]

### Expected behavior

1. run the Soundboard app
2. Be able to manually set frequencies using typing method and play any sound .wav files 8 or 16 bit unsigned or signed mono files between 24000 and 48000 Hz

### Environment/versions

Mayhem v2.0.2 and SDCard

AKProwl, 2024-09-23:
I've reduced the amount of sound .wav files to 411MB. And now I can seemingly run it fine now

zxkmm, 2024-09-23:
It should be the list which ran out of all the ram, not quite related to the file size if the app works correctly.

There's not much that we can do because it's kind of hardware limitation.
It's maybe possible to use the file picking view which HTotoo already did the compatibilities work for low ram, but in that case you can only pick and play files one by one, can't be a playlist like what we have currently.

I think you can make multiple wav directories (WAV1 WAV2 WAV3) and rename among of them (to WAV) as a workaround. and wait for better idea.

htotoo, 2024-09-23:
There is already a paginator but with 100 item / page limit. I reduced it to 50 / page, so it may be good enough.
Linked the test ppfw. can you flash it, and test if it is working for you with all the files (without creating sub directories, ..)?
Thanks.

https://cdn.discordapp.com/attachments/722101917135798312/1287746002752634962/portapack-mayhem_OCI.ppfw.tar?ex=66f2aa6c&is=66f158ec&hm=6effb78e48cb5f51d4c021f22bc943dbfd09a157fee3814113996b62f5639c52&

closed 2024-09-28 by htotoo