portapack-mayhem/mayhem-firmware #2264: No world_map.bin file in /ADSB/ Directory

HamIam77, 2024-09-24:
### Describe the bug.

With the SD card installed attempting to open the map in ADS-B I get this message: No world_map.bin file in /ADSB/ Directory 
When I open the folder on the SD card the file is there. Please help, one of the cooler features. 
[image]

### Reproduction

Adsb mode. Click on plane. Track.
[image]

### Expected behavior

To see a map with GPS location of an airplane 

### Environment/versions

New today. Latest firmware 

### Anything else?

Please Help

Leiflet, 2025-01-03:
i have done that yet i still get that message

htotoo, 2025-01-03:
You put files.to.thenwrongnplace.
They needs to be.in the root of the sd..

MikayelMa, 2025-03-07:
I have doe everything correctly. Files on the SD root, firmware updated to the latest version and yet it says that no World map Bin is in the ADSB file. what should I do?

Eltiefao, 2025-07-11:
Também estou com esse problema acho que pode ser a versão do HackRF uso um H2 r9 recebi ele ontem e quando coloco essa pasta de mapas no cartão e coloco no RF aparece essa mensagem e todos os arquivos das outras pastas desaparecem

Eltiefao, 2025-07-11:
I'm also having this problem, I think it might be the HackRF version. I use an H2 r9, I received it yesterday and when I put this map folder on the card and put it in the RF, this message appears and all the files in the other folders disappear.

Eltiefao, 2025-07-11:
I went to update my RF h2 hack and it got stuck on the black screen, does anyone know how to solve it?