portapack-mayhem/mayhem-firmware #2275: Ble spam select device spam

theguythatdidsomething, 2024-09-29:
### Description of the feature you're suggesting.

Select what to spam like AirPods not random

fr1sk, 2024-10-09:
@theguythatdidsomething I can implement this, but let's first discuss how the user will choose what to send. We already have the random device button, so perhaps if iOS is selected, instead of the random device checkbox, we could display a left/right selector. The default option could be `rnd device`,  but scrolling left or right would allow the user to select specific devices like AirPods, Apple TV, a new phone, etc. @jboone what do you think?

theguythatdidsomething, 2024-10-09:
> default

I love the idea it was just what i was thinking

htotoo, 2024-10-09:
> @theguythatdidsomething I can implement this, but let's first discuss how the user will choose what to send. We already have the random device button, so perhaps if iOS is selected, instead of the random device checkbox, we could display a left/right selector. The default option could be `rnd device`, but scrolling left or right would allow the user to select specific devices like AirPods, Apple TV, a new phone, etc. @jboone @eried @htotoo what do you think?

Seems good, as long as it can fit the ext app filesize and memory usage. Feel free to implement that!
We'll see the details on the PR :)

theguythatdidsomething, 2024-10-09:
Last little question if this does get implemented how long would it take to be in a update

No pressure just curious

gullradriel, 2024-10-09:
If you implement it, it would make you a contributor, and in that case, the time it would take would depend on you. 

Giving an idea and not doing the code => not making you a contributor.

If we are making it (@htotoo  or anyone), then it will be ready when it will be ready. 

Once ready it's generally quickly implemented in 'next' / available in the nightly following the merge.

And it would be in stable when we make a stable release. No dates for this, it's when we think we have enough new working things / fixes.

htotoo, 2024-10-09:
If it is urgent, you can use the BLE tx app. You just need to have a txt file with the ble packet in it. I think there are some available, so you can ty and copy.

fr1sk, 2024-10-09:
@theguythatdidsomething I should be able to deliver this in the next couple of days (most likely over the weekend). However, I can't guarantee it, as this is my first contribution to Mayhem, and I need some time to get familiar. I'm also balancing this with my regular work, so it's being done in my free time.

theguythatdidsomething, 2024-10-09:
> If you implement it, it would make you a contributor, and in that case, the time it would take would depend on you.
>
> Giving an idea and not doing the code => not making you a contributor.
>
> If we are [...]

it sounds like i got you mad sorry if so i was just asking

gullradriel, 2024-10-09:
> it sounds like i got you mad sorry if so i was just asking

Certainly not. You were asking, I answered.

Trust me, if I had gotten mad, you would clearly have noticed it, and it would not be an impression ;-)

htotoo, 2024-10-09:
> @theguythatdidsomething I should be able to deliver this in the next couple of days (most likely over the weekend). However, I can't guarantee it, as this is my first contribution to Mayhem, and I [...]

Feel free to join discord and ask if you got any questions. But keep in mind, decs are from all around the world, so maybe have to wait for an answer . But anyway it worth to join :)

theguythatdidsomething, 2024-10-09:
> from

Thanks for the suggestion but i am already on there quite an active user in fact

theguythatdidsomething, 2024-10-10:
@fr1sk  i have tried working with the code no luck any luck by your side

eried, 2024-10-11:
> @theguythatdidsomething I should be able to deliver this in the next couple of days (most likely over the weekend). However, I can't guarantee it, as this is my first contribution to Mayhem, and I [...]

Did you decide on the UI part to select a device? I think the arrow is the one that makes sense since is like that on all the other apps

theguythatdidsomething, 2024-10-11:
Yes what you say is the best idea but lets also  include a button that says select variant like green airpods max not the normal white ones so its on  you but if posible we cant put that 2 button

theguythatdidsomething, 2024-10-11:
> Did you decide on the UI part to select a device? I think the arrow is the one that makes sense since is like that on all the other apps

es what you say is the best idea but lets also include a button that says select variant like green airpods max not the normal white ones so its on you but if posible we cant put that 2 butto

htotoo, 2024-10-11:
Remember, the code is simplified a lot, so it'll be a big task, to integrate the missing code, and also let it fit the  flash. And also the new controls will take space, and the options, ..
Iirc most of the colors are removed, ...
So the device selection could work, but the sub variant I don't think

theguythatdidsomething, 2024-10-11:
> Remember, the code is simplified a lot, so it'll be a big task, to integrate the missing code, and also let it fit the  flash. And also the new controls will take space, and the options, ..
>
> Iirc [...]

Thats why i said if possible

fr1sk, 2024-10-14:
> Remember, the code is simplified a lot, so it'll be a big task, to integrate the missing code, and also let it fit the flash. And also the new controls will take space, and the options, .. Iirc most of the colors are removed, ... So the device selection could work, but the sub variant I don't think

> Did you decide on the UI part to select a device? I think the arrow is the one that makes sense since is like that on all the other apps

@htotoo @theguythatdidsomething exactly, it will be pretty hard to add everything and to fit in to the flash. @eried I am starting with the arrow select of the device. I made some progress but a bit less than expected, but Im aiming to open the PR in next couple of days.

theguythatdidsomething, 2024-10-14:
> @htotoo @theguythatdidsomething exactly, it will be pretty hard to add everything and to fit in to the flash. @eried I am starting with the arrow select of the device. I made some progress but a bit [...]

What is pr

theguythatdidsomething, 2024-10-15:
> > > > Remember, the code is simplified a lot, so it'll be a big task, to integrate the missing code, and also let it fit the flash. And also the new controls will take space, and the options, .. Iirc most of the colors are removed, ... So the device selection could work, but the sub variant I don't think
> 
> > > 
> 
> > > 
> 
> > > > Did you decide on the UI part to select a device? I think the arrow is the one that makes sense since is like that on all the other apps
> 
> > > 
> 
> > > 
> 
> > > @htotoo @theguythatdidsomething exactly, it will be pretty hard to add everything and to fit in to the flash. @eried I am starting with the arrow select of the device. I made some progress but a bit less than expected, but Im aiming to open the PR in next couple of days.
> 
> > 
> 
> > What is pr
> 
> 
> 
> PR is Pull Request, which will have a feature of selecting the devices

So it's ready?

htotoo, 2024-10-15:
Please stop spamming. Wait for it with patience.

theguythatdidsomething, 2024-11-05:
Sorry for spamming I promise this is my last post until it comes out so I congratulate you but any updates on it @fr1sk

fr1sk, 2024-11-06:
@theguythatdidsomething sorry for taking this long, but had some personal things to take care of. I will give you some short update:
I manage to include colors dynamically for each device as well. While testing, no matter what I sent I always get default color. Seems like the codes we have for colors for specific device are not working, so I am currently trying to figure out what part of the package is missing/wrong to be able to present color correctly in the iphone popup. 

Once I fix this, it will be published, but I cannot give you the ETA since I don't know how long it will take to figure it out.

theguythatdidsomething, 2024-11-06:
@fr1sk its okay health and personal life always goes first I myself have been doing  the same as you I can't figure out what makes the device a different color it's supposed to be some digit in HEX but I haven't figured wich one I'll update you if I find anything take care

theguythatdidsomething, 2024-11-07:
> @theguythatdidsomething sorry for taking this long, but had some personal things to take care of. I will give you some short update:
>
> I manage to include colors dynamically for each device as well. [...]

Could you send me the color codes to do some testing myself

fr1sk, 2024-12-09:
I'm a bit stuck here 😞 I successfully managed to dynamically load colors for each device and select both the device and its color from the menu. However, no matter which color option I send, it always defaults to the original color. If anyone has a device that doesn't use the default colors or any ideas on how we can debug this, I'd really appreciate your help. Thanks!

@htotoo @gullradriel @theguythatdidsomething

theguythatdidsomething, 2024-12-09:
> I'm a bit stuck here 😞 I successfully managed to dynamically load colors for each device and select both the device and its color from the menu. However, no matter which color option I send, it [...]

I have been working on it to I cannot make do it sends the devices with different colors y modify the packet but I recive normal color let's just release it and then we work on that problem