portapack-mayhem/mayhem-firmware #2285: All nightly firmwares after V2.0.2-stable crash the portpapack H1

ebs-10, 2024-10-06:
### Describe the bug.

All nightly firmwares after V2.0.2-stable crash my portpapack H1. 
It boots to the HACKRF Portapack Mayhem splahscreen (also the hack.app still works then).
If you then press the middle button to go to the menu, the firmware (in this case Nightly Release - 2024-10-06) crashes with error: 

M0 Guru Meditation 
Hint: hardware Error
See screenshot for all details
I also added a pdf with screenshots of the  the DFU stack dump.
I can revert back to V2.0.2 stable and this works without any issue.

I tried Nightly release 2024-10-06 back to 2024-09-26.
[DFU.pdf](https://github.com/user-attachments/files/17268687/DFU.pdf)
[M0.pdf](https://github.com/user-attachments/files/17268688/M0.pdf)

### Reproduction

Connect HackRf with Portapack H1 to hackrf.app
Upgrade to latest nigthly (2024-10-06)
Reboot the device
Starting up with splashscreen
Press middle button to go to the menu

### Expected behavior

After pressing the middle button, it should go to the menu screen of Mayhem.

### Environment/versions

All nightly firmwares after V2.0.2-stable

gullradriel, 2024-10-06:
Cannot reproduce on my own H1.
Please follow this:
- clean your sdcard, re add contents from https://github.com/portapack-mayhem/mayhem-firmware/releases/download/nightly-tag-2024-10-06/mayhem_nightly_n_241006_COPY_TO_SDCARD.zip
Then follow https://github.com/portapack-mayhem/mayhem-firmware/wiki/Factory-Defaults

Most of the time this is happening because you are not waiting for the device to reboot BY ITSELF at the end of flashing.

zxkmm, 2024-10-06:
could you pls try if touch the screen in splash works? 
could be this but it shouldn't , since i didn't change button press handle logic. only touch screen. https://github.com/portapack-mayhem/mayhem-firmware/pull/2260

zxkmm, 2024-10-06:
also delete custom splash (if you are using) and try again

zxkmm, 2024-10-06:
I don’t think there’s portapack without touchscreen. 
Would you mind post a video of the backside of your portapack?

and are you sure it says “hardware error”? 

also could you pls use a firmware that doesn’t have issue when booting, and enter debug menu and test those functions that related to touchscreen?

zxkmm, 2024-10-06:
Also “without touchscreen” you meant without a screen, or with screen but not touchable? (I can identify it if you post a picture of your screen (without metal case, plastic case would be ok)

and also take out sdcard and button batter (coin batter) and try again

gullradriel, 2024-10-06:
@ebs-10 we may need photos of the portapack and hackrf board to see if it's a strange edition
Would you be OK to dismantle the whole thing and send us pictures of each side of each board (without enclosure/battery/internal speaker hiding the board) ?

ebs-10, 2024-10-06:
The screen is not a touchscreen. After my purchase, i asked the seller why the touch wasn't working. The reply was that this version didn't have touch. This is confirmed when i tried to calibrate the screen. So this version of the portapack only has a screen without touch. This would explain the problem after the alteration of the firmware i guess.
I've added some pictures of the boards.

[image]
[image]
[image]
[image]
[image]

gullradriel, 2024-10-06:
Thanks for taking the time to dismantle and send us photo, I hope it will help us to fix it or state if it's no more fixable.

ebs-10, 2024-10-06:
> Thanks for taking the time to dismantle and send us photo, I hope it will help us to fix it or state if it's no more fixable.

Is it possible to make #2260 selectable (in a ini file) so we can enable or disable this feature through the SDcard? This way we can rule out if this is the problem.

zxkmm, 2024-10-07:
We are not yet sure if it’s from #2260 . Just suspect but it doesn’t make sense

zxkmm, 2024-10-07:
Just confirmed it’s not from #2260 , since 2.0.2 already included it and you said it works good

ebs-10, 2024-10-07:
> Just confirmed it’s not from #2260 , since 2.0.2 already included it and you said it works good
Yes, that is correct.

- I just traced back all the nightly firmware starting from september (Nightly Release - 2024-09-08).
- All nightly firmware' s worked up to  n_240923 - nightly (worked)
- The problem started with nightly n_240924 - nightly (failed with the same error as originally posted)
Strange thing is that V2.0.2 works fine while #2260 in included. 
Now i'm back on v2.0.2 without a problem....

[image]
[image]

zxkmm, 2024-10-07:
Oh wait ya sorry 2.0.2 is actually before 240924, it seems GitHub have some issue displaying the order.

could you pls go to Debug - Button Test and also Touch Test to see if they crash?

ebs-10, 2024-10-07:
Hi,
Here is a video of the button test en touch test (which is not available on this LCD screen)
https://github.com/user-attachments/assets/d57799d2-ca61-4330-8329-ad36851929a2

zxkmm, 2024-10-07:
thank you very much for the video, I made a custom firmware package, if you don't mind, please flash it (with hackrf.app AKA mayhem hub, use the button following:), and do follow:

[image]

1. if crashed: take a picture of the red outline screen firstly, and then press DFU button and use up and down buttons to show me all address that shows on the screen.  (plz make sure the image is from this specific firmware, because all the address is different among of all the firmware versions) (rename and remove the .PDF extension firstly to get the .tar file (so you can flash wth mayhem hub) because github won't allow me to upload tar files)

[portapack-mayhem_OCI.ppfw.tar.PDF](https://github.com/user-attachments/files/17281127/portapack-mayhem_OCI.ppfw.tar.PDF)

2. if not crashed: lemme know and i'll send another test firmware.

zxkmm, 2024-10-07:
also pls join our discord server, because use github to chat is a little bit laggy and limited.
the invite link is on the main page of this repo

ebs-10, 2024-10-07:
> thank you very much for the video, I made a custom firmware package, if you don't mind, please flash it (with hackrf.app AKA mayhem hub, use the button following:), and do follow: [...]

This firmware works like a charm. so it looks like you found the issue?
I'll install discord tonight.

zxkmm, 2024-10-07:
you don't need to install it, just use https://discord.com/app , personally i had never install discord

htotoo, 2024-10-07:
You can use discord from the web, no need to install! What about the newest nightly? still bad? 
how you start the portapack? is the usb plugged in when you turn it on, ore not? if yes, to a computer, or to just a power source?

zxkmm, 2024-10-07:
also sorry for the thing but pls test this one (should crash) and show me all the address like i asked above
[portapack-mayhem_OCI.ppfw.tar.PDF](https://github.com/user-attachments/files/17281668/portapack-mayhem_OCI.ppfw.tar.PDF)

ebs-10, 2024-10-07:
Here is the requested file for the testfirmware you send me (the latest portapack-mayhem_OCI.ppfw.tar Firmware Version: e3facb0a).
This one crashed as you predicted.
I start by connecting to the PC or an external powerbank (with 2.1A). With both the firmware crashes. 
Tested n_241007, this firmware also crashes On USB via PC and External powerbank and also and external power supply of 65w.

[Scan - 2024-10-07 18_43_51.pdf](https://github.com/user-attachments/files/17282059/Scan.-.2024-10-07.18_43_51.pdf)

ebs-10, 2024-10-08:
Issue resoved by disabeling the touchscreen under 
- Settings, User Interface
- option ' Disable touchscreen' had a red cross, so was disabled.
- I set it to enabled (under working firmware V2.0.2) and save the config. 
- Then i flashed the former faulty firmware n_240924 and it worked. 
- After that i flashed the latest n_241008 also without a problem.
- So the setting was the problem.
[image]
[image]

closed 2024-10-16 by zxkmm