portapack-mayhem/mayhem-firmware #2292: compile-flash-mayhem-deb.sh doesn't complile earlier versions

dudelaksdfjlkaz0102, 2024-10-12:
### Describe the bug.

compile-flash-mayhem-deb.sh
compiles fine for later versions
but not v1.4.0 that's helpful for replay of .c16 files

Error messages received:
GENHDR include/libopencm3/stm32/l1/irq.yaml
FILE "./scripts/irq2nvic_h", line 32, in <module> import yaml
ImportError: No module named yaml
make: *** [Makefile:103: all] Error 2

### Reproduction

Reproduce by modifying:
compile-flash-mayhem-deb.sh
adding -b v1.4.0 to the following line:

sudo get clone -b v1.4.0 --recurse-submodules https://github.com/eried/port a packs a them.git

### Expected behavior

I expected compile-flash-mayhem-deb.sh to compile v1.4.0 as it compiled other versions using -b [version]

### Environment/versions

Debian 11

zxkmm, 2024-10-12:
I don't think this script exist in this repo. Where did you get it? And can you try following with the wiki to compile?
Thank you.

gullradriel, 2024-10-12:
It's mine.
Please open the issue there:
https://github.com/gullradriel/compile-flash-mayhem
Thanks

closed 2024-10-12 by gullradriel