portapack-mayhem/mayhem-firmware #2310: Settings reset when unplugging from power/USB

DrewMeager, 2024-10-18:
### Describe the bug.

I have the Portapack H2 and was having trouble updating to Nightly release - 2024-10-16 from Nightly release - 2024-10-15. I decided to wait until a new nightly was released in case it was a bug that was known so when I saw that Nightly release - 2024-10-18 had been released, I decided to give it a shot. it updated just fine, but now anytime I unplug it from USB or even just power it resets the UI settings. For example, I have it set to not show the plash screen on start-up as well as disabling some of the controls on the top bar (same area as the brightness control)

### Reproduction

1. plug into USB/power
2. unplug. (it can be powered on or off, it does not matter)

### Expected behavior

UI settings will reset (splash screen displays on start-up and controls are re-enabled)

gullradriel, 2024-10-18:
You coin battery is dead.

gullradriel, 2024-10-18:
Change the coin battery, or use pmem to sdcard (in settings) to save your settings on the sdcard.

DrewMeager, 2024-10-18:
I feel so stupid for not even thinking about the coin cell. I will give that a shot and let you know. thank you

gullradriel, 2024-10-18:
No problem :-)
Tell us if it's finally working, so that we can close the issue.

closed 2024-11-05 by gullradriel