portapack-mayhem/mayhem-firmware #2314: Audio issue (no audio,no volume change) on Hackrf portapack H2+ (pcb v3.4b) with v2.0.2 - stable

Darkmatter3141, 2024-10-19:
### Describe the bug.

Hello,after ugrading to 2.02 audio issues comes up.Sometimes in debug menu ,audio chip is recognized as WM8731 and sometimes as AK4951.
I tried to remove coin battery and the 3,7v pack,also sd card was pulled.

Audio test doesn't produce audio tone in any case.
After downgrade to v 1.2 ,audio start works ,audio test was fine but audio level was at fixed level (doesn't matter if value change from 0 to 99).
After upgrade to 2.02 audio stops,
Upgraded to n_241018 audio problem persist.
Green icon for audio enabling always green in all previous tests.
Hardware ispection for poor solder ,or broken pads ,on 3.5mm audio jack was done and no issue found.
Played again with audio test (under debug menu) and setted duration (ms) to a value greter thn 5000 for several times....audio start working also on other apps untill reboot...but audio level never change if value are setted from 0 to 99.
   

### Reproduction

Not sure about causes,i cannot exclude an hardware problem but....after several audio tests  from debug menu, start works untill reboot. 
Audio tests was made modifing duration (ms) from 10 to 6000 and audio volume from 0 to 99.
Audio Icon start test seems work reversed.... i can hear audio test when is grey and no audio is present when is green. 

### Expected behavior

If audio is not available....playing with audio test solve,temporary,the issue.

### Environment/versions

Mayhem portapack v2.0.2 - stable

### Anything else?

Seems a lack of audio device initialization during some boot.Maybe the audio test re initialize the device.

gullradriel, 2024-10-19:
When upgrading for such large different versions, you should perform a factory reset.
Reset persistent memory, reset applications files settings, and use last up to date Sdcard contents.
And if you want us to help, you have to stop downgrading to something so old as v1.2
Please also provide pictures of your hackrf and portapack boards, disassembled and both faces.
Thanks :-)

gullradriel, 2024-10-19:
Hah, and when upgrading to anything like last stable or nightly, be sure to use corresponding sdcard content, as a lot have changed from old 1.2 and 2.X series.

gullradriel, 2024-10-19:
(I didn't want to assign @ArjanOnwezen , it was a bad click when scrolling up and down. That message is just for him to know, in case he already got the ping)

gullradriel, 2024-10-19:
Also please check that your coin battery is not dead, and if it's the case change it or use pmem to Sdcard.

ArjanOnwezen, 2024-10-20:
( Hi @gullradriel long time ago :) )
@Darkmatter3141 just use another SD card with fresh content. To make sure it matches the version of the firmware.

Darkmatter3141, 2024-10-20:
Hello,
Thanks for your reply.
I started from the beginning as you advice:
Removed and used a new sd card (sandisk utra 32Gb class 10) ,loaded on sd card "mayhem_v2.0.2_COPY_TO_SDCARD.zip
" (extracted),loaded new firmware from sd card and factory reset done (yes...version on main page is now v2.0.2). 
(settings->P.Memory Mgmt->Reset P.Mem to defaults ....done 3 times for safety). 
Are there any other method to do a factory reset ?

Coin battery was replaced one day ago but ,for safety i measure it ...found 3.161V (not so bad).
Battery pack is full charged and mantain for at least 20 minutes.

I understand your advice to not downgrade to very old version but ...this was only a test to avoid hardware
issues ( the web page for remote view of hackrf is fantastic and i don't want lose it).

After all, audio is still tricky...audio volume seems not follow 0-99 values (speaker icon is green) and i 
still have to do Audio beep test procedure 2-4 times before audio starts to work.

I suppose be my hardware problem...no one seems be affected (fortunately).

Here some photos....i hope they are right what you asked.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

gullradriel, 2024-11-05:
Checking against the boards I have, it looks like my own H2 device.
I'm sorry to tell you that I don't have any of the mentioned problems you have.
It's strange you need more than one press on the audio tests to have something.

Last resort checks:
- check the audio cable you are using, try another one
- check the audio input you are using to get back the sound, the drivers of the sound cards, it's levels
- try to use simple headphones
- check the soldering of the jack

q9droid, 2024-12-04:
My new portapack h4m also has no sound. Version 2.02. No sound in headphones and speaker

gullradriel, 2024-12-20:
Closing due to inactivity.
@q9droid open a new issue if you are still having the problem. You should contact the reseller as we did not touch the audio part.