portapack-mayhem/mayhem-firmware #2317: I removed power while flashing

monerotop, 2024-10-22:
### What happened?

hi, when i was flashing inside hackrf, i saw that its not reacting, after i removed power and on it again, its having black screen, all leds on, and power percent are working, i was trying to re flash in dfu mode but hackrf tool not seeing it, i also tried dfu-util, it was showing it, but when flashing its giving me error.

sudo dfu-util -d 1fc9:000c --download /Users/monero/Downloads/mayhem_nightly_n_241021_FIRMWARE/portapack-h1_h2-mayhem.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1fc9:000c
Device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[                         ]   0%            0 bytesdfu-util: Error during download (LIBUSB_ERROR_IO)

[image]
[image]

### Your hardware

Hackrf 

### Tests

- [X] I managed to update/boot previously

eried, 2024-10-24:
I only know the Windows part, but with that error I would use zadig (if you have access to a Windows machine) check that in dfu you get "lpc" device, instead of something dfu or hackrf... In this case open zadig and asign lpc to the default driver (check the documentation, don't remember exactly which one). And now dfu with fix your device, without rebooting install mayhem

gullradriel, 2024-11-11:
Hello,

Please read those, and come back open an issue if you are still having a problem:

https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware-troubleshooting
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Diagnose-firmware-update-in-Windows
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Troubleshooting
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Updating-the-Xilinx-CPLD-on-hackrf-board
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Factory-Defaults

Thanks.

closed 2024-11-11 by gullradriel

monerotop, 2024-11-12:
> I only know the Windows troubleshooting: 
> 
> 
> 
> With that error I would use zadig (if you have access to a Windows machine) check that in the Device manager you get "lpc" device, instead of something dfu or hackrf... In this case open zadig and asign lpc to the default driver (check the documentation, don't remember exactly which one). 
> 
> 
> 
> Now do the dfu fix, and without rebooting install mayhem

Thank you very much, I entered in dfu, after installed driver from drivers folder, after I opened file dfu_hackrf_one.bat, after in device manager I saw hackrf, after that I opened flash_portapack_mayhem.bat and it worked