portapack-mayhem/mayhem-firmware #2323: Delay in loop iteration

dudelaksdfjlkaz0102, 2024-10-28:
### Description of the feature you're suggesting.

I'd like to add an option for replay app to sleep between loop iterations.

I'd like to see the same GUI used to adjust signal gain for replay app to adjust delay between loop interations to be 0-60 minutes added next to loop check box.

I assume it could be as easy as copying and pasting current code, renaming things in pasted code and adding a sleep command.

### Anything else?

I'd like to add this feature myself. Advice for adding this feature? I'm able to compile latest source code to replay my signal.

I'm not certain if I will be making changes to files other than:

mayhem-firmware/firmware/application/
replay_thread.cpp
replay_thread.hpp

mayhem-firmware/firmware/application/apps
replay_app.cpp
replay_app.hpp

mayhem-firmware/firmware/baseband
proc_replay.cpp
proc_replay.hpp

Any advice to make this modification?

htotoo, 2024-10-29:
the replay app is in the ui_playlist.cpp.
there is a message that is sent when the current tx is done, and that calls a this function:

`void PlaylistView::handle_replay_thread_done(uint32_t return_code) {
    if (return_code == ReplayThread::END_OF_FILE) {
        if (next_track()) {
            send_current_track();
            return;
        }
    }

    if (return_code == ReplayThread::READ_ERROR)
        show_file_error(current()->path, "Replay read failed.");

    stop();
}
`

this is a part, where you could start a timer (maybe based on rtc_time::signal_tick_second ), and if it reaches enough, then send the next track (if any).
but this is only good, when you need seconds based resolution.

zxkmm, 2024-10-29:
> I'd like to add an option for replay app to sleep between loop iterations.

This is already implemented long ago, check wiki. But feel free to improve it (currently it freeze when pausing)

gullradriel, 2024-11-03:
> I assume it could be as easy as copying and pasting current code, renaming things in pasted code and adding a sleep command.

I'm waiting for your Pull Request :-)

zxkmm, 2024-11-11:
After reading this again I noticed that this actually already implemented, just use playlist, set your timeout, add tracks that you want (or only add one if you want to loop one)
Then check loop checkbox, load and play the playlist, it will work as you wanted.
I'm closing but feel free to re-open if this isn't working as you expected.

closed 2024-11-11 by zxkmm