portapack-mayhem/mayhem-firmware #2324: RS41 -> what have been changed in radiosonde module from FW v2.0.1 to v2.0.2?

voxo22, 2024-10-28:
### Describe the bug.

Log of Radiosonde RS41 no more readable by RS41Tracker software.
There has been something changed in code, I cannot load content of log SONDE.TXT to RS41Tracker software. Log of the same sonde with FW version 2.0.0 and 2.0.1 works fine.

### Reproduction

It cannot be reproduced without my special conversion script

### Expected behavior

same as in FW v2.0.1

htotoo, 2024-10-29:
could you provide a good and a bad log please?

voxo22, 2024-10-29:
[SONDE-bad.TXT](https://github.com/user-attachments/files/17555881/SONDE-bad.TXT)
[SONDE-good.TXT](https://github.com/user-attachments/files/17555883/SONDE-good.TXT)

htotoo, 2024-10-29:
can't load the bad or the good file on the app (i may use it badly), happens the same with both files.
checked the changelog, and nothing related to sonde app changed.

could you check with a new nightly build?

voxo22, 2024-10-29:
ok, you cannot load it directly, RS41tracker use different syntax of data. You have to:
1. remove timestamp and space on each row
2. separate bytes by space (e.g. 93 A2 51...)
3. add byte prefix "86 35 F4 40 " on start of each row
4. save

Then you can smoothly load datafile to tracker (option load raw data file).
The bad one will be completely ignored (no calibration point = yellow dot on lower left corner will be decoded)

I made log immediately one after another from the sonde yesterday, fully functional. Also tested with previous FW version 2.0.0 without any problem.

gullradriel, 2024-11-05:
Not that much have changed between both versions.
Can you check with a nightly ?
Or with, in example, the last up to date build of the day in #test-drive ? ( https://discord.com/channels/719669764804444213/722101917135798312/1303347923207323670 )

voxo22, 2024-11-05:
@gullradriel yep, the last up-to-date build is OK!  FW2.0.2 is broken.

gullradriel, 2024-11-05:
The use the nightly until next stable is out ;-)
Thanks for your tests.

closed 2024-11-05 by gullradriel