portapack-mayhem/mayhem-firmware #2325: Frequencies transmitted by signal generator and mic app are incorrect or missing above 1ghz - 1.3 ghz.

chris451gh, 2024-10-29:
### Describe the bug.

With firmware 241018 signal generator is not on the correct freq above approx 900mhz. Set signal gen app to 1288, observe a signal on 1268. Use the dial to tune up. The displayed freq increments up as expected but the output frequency does not match but may be within 20mhz of the indication.  Observe with spectrum the frequency jumps around several mhz.
Switch to the TX app. Observe similar frequency errors. As it is tuned with 10khz steps it jumps to serveral mhz away.
I have observed reboots when the TX is pressed, that might be current limitation of the USB port.
The HackRFportapack was  reloaded with firmware 2.01 and the signal generator and transmit app are generating the correct frequency..
Spot checking indicates signal generator outputs at 800mhz correctly. Not true of 1001 mhz. 900.000 is OK and entering 1001 mhz then pressing start causes a reboot.
 I have two hack RF portapacks and both got this problem with firmware 20241018. I verified no problems with Hackrf mode on these freqs. No problem with v2.01 release. I have transcievers in the 1240-1300 band and they were not picking up the hackRF.

### Reproduction

Load firmware 241018. Utilities>Signal genrator. Set frequency to 145.5mhz and check the signal.
Repeat at 900 and verify a signal. Now enter 1001mhz and press start...This unit crashes restarts.
Enter 1280mhz. Press start. Check for signal - non found. Try this with later versions I have not.
Tuning to a receive freq in the 1280-1290 range and in Receive Audio Spec mode the signal is not received where its expected.

### Expected behavior

A signal should be found with a spectrum analyzer or receiver on the specified freq.
Check for an error after version 2.01 which works correctly RX TX and SigGen. 2.01 was verified up to 6 ghz to cover all bands correctly. 

### Environment/versions

2024-10-18 firmware.   2.01 operates normally for this test.

### Anything else?

I keep many versionis of firmware on the SD card. When going back to 2.01 I did not change any apps, left them as they were supplied in the 20241018. If this was detected in later versions please make a note in the version logs.

chris451gh, 2024-10-29:
When I reloaded 2.01 it was from the .tar file on my SD so it did replace some of the apps for that version.
I am using the HackRF.app to view and update the HackRF portapack (H2 version).

zxkmm, 2024-10-29:
I cannot reproduce your issue by:
Go to 145.5, looks correct
Go to 900, looks correct
Go to 1280, looks correct
Use dial to fine tune the freq, looks correct
Feel free to correct me if I read something wrong, thx!

zxkmm, 2024-11-24:
close as no future information provided from submitter for very long, and also not able to reproduce the issue.
Feel free to re-open again if you or anyone provide more info.
thank you!