portapack-mayhem/mayhem-firmware #2329: there is many spurious emission when replay lora(C16) signal.

ruanxw, 2024-10-31:
### Describe the bug.

hardware: portapack H2(MAX2837 SI5351C)
firmware: V2.0.1

lora iq file(52.3k_62.5k.c16)
   bandwidth: 52.3khz
   sample rate: 200k
   url: https://drive.google.com/drive/folders/1mRJ-CC0OTODt6pOXMl4QfbpQ2Tush09h?usp=sharing

i replay this file by portapack H2 and chek the wave by r&s device pr200. there are many spurious emission. you can check the attached picture(testResult.png).

[image]

**The sample rate is 200khz. And we can find that every 200khz there is a spurious emission.** 

Could you please help to check it? Thank you so much!

### Reproduction

1. Switch on
2. enter replay app.
3. select the lora iq file lora_52.3k_62.5k.c16
4. start to replay.
5. check the signal waveform with receiver（R&S PR200)

### Expected behavior

1. the signal wave is good without spurious emission.

ruanxw, 2024-10-31:
we want to replay many signal(FM/AM/QPSK/LORA...) to study. But all Spectrum are incorrect.

zxkmm, 2024-10-31:
I didn't check your iq file but from the spectrum, it looks mostly like echo or bad antenna-related factors, like bad coupling and/or wrong/bad band-pass filter in receiving device.

That saying, this is mostly from hacked/antenna/receiver, not quite portapack related.

By the way, I noticed that you are probably from a university in china, just saying that if you are a professor or tutor and can get me an admission offer of postgrad school, I probably can provide more help.

ruanxw, 2024-10-31:
1.sorry, i don't work in university. 
2.We also test with gnuradio in unbuntu. Every thing is same except the SDR device. One is portapack in hackrf mode and another is uhd B210.  B210 is good. But portapack is same as previous test.

ruanxw, 2024-10-31:
> I didn't check your iq file but from the spectrum, it looks mostly like echo or bad antenna-related factors, like bad coupling and/or wrong/bad band-pass filter in receiving device.
>
> That saying, [...]

1.sorry, i don't work in university. I just study by myself.
2.We also test with gnuradio in unbuntu. Every thing is same except the SDR device. One is portapack in hackrf mode and another is uhd B210. B210 is good. But portapack is same as previous test.
So do you have anyother suggestion?  Hope your reply. Thank you so munch!

zxkmm, 2024-10-31:
what software you use to replay the iq?

ruanxw, 2024-10-31:
> what software you use to replay the iq?

test two times:
1, portapack h2 replay app(v2.0.1)

2, file source in gnuradio 3.10 in pc(ubuntu)
[image]

by the way, what's the max value of sample rate that portapack h2 replay app support?  Someone told me that the sample rate should be higher than 8MHz.

zxkmm, 2024-10-31:
please try this file 
[lora3_52.3k.complex.zip](https://github.com/user-attachments/files/17588261/lora3_52.3k.complex.zip)

ruanxw, 2024-10-31:
> please try this file [lora3_52.3k.complex.zip](https://github.com/user-attachments/files/17588261/lora3_52.3k.complex.zip)

should i try with replay app in portapack h2 or gnuradio in pc?  The sample rate is 200k?

ruanxw, 2024-11-01:
I replay this file by gnuradio in pc. These spurious emission still persist.

[image]
[image]

zxkmm, 2024-11-01:
And what if you replay your original c16 file using the replay app in portapack?

zxkmm, 2024-11-01:
Also show me your antenna

zxkmm, 2024-11-01:
And if you tried both way (hackrf+computer and hackrf+portapack)
And they aren’t work, then it’s not from here. Maybe open a ticket on hackrf repo.

ruanxw, 2024-11-01:
actually i replay  the original c16 file in both way (hackrf+computer and hackrf+portapack). And test your data file with  hackrf+computer. The bug always persist. So it should be hackrf bug.

And i have last question. What is the max sample rate that portapach H2 replay APP supports?

ruanxw, 2024-11-01:
[image]

zxkmm, 2024-11-01:
I assume 5.5MHz but I don’t know if there’s hard limit. Maybe others know.

closed 2024-11-01 by zxkmm