portapack-mayhem/mayhem-firmware #2330: Feature request

Ademolght, 2024-10-31:
### Description of the feature you're suggesting.

i want a ble jammer for portapack h4M if it's not avaible

zxkmm, 2024-11-01:
It’s not quite possible, since the max bandwidth is still too narrow to cover the freq range of BLE.

and if you try hopping the freq to cover, BLE is designed to defense this situation (signal interference)

zxkmm, 2024-11-01:
As mentioned by another developer on mayhem discord, this could be possible, so I'm reopening this and we'll dig deeper and see what could happen:)

Ademolght, 2024-11-02:
> As mentioned by another developer on mayhem discord, this could be possible, so I'm reopening this and we'll dig deeper and see what could happen:)

ok np