portapack-mayhem/mayhem-firmware #2334: When will the Portapack H2 release SSTV reception and NOAA decoding functionality?

324569375, 2024-11-02:
### Description of the feature you're suggesting.

I found that the Portapack H3 has the capability to receive SSTV and decode NOAA signals, as well as provide train alerts and other functionalities. However, the Portapack H2 does not currently have applications for these features. I hope that the Portapack H2 will also receive updates to include these functionalities, and I am also looking forward to the implementation of analog TV transmission capabilities. Once again, I would like to thank the developers for their efforts and contributions to this project.

gullradriel, 2024-11-02:
Please contact your H3 reseller and ask him why he is not pushing back the code modifications he did.
Because that's what is happening.
We are not making a firmware for a specific version. We are making a firmware for all versions.
The guy selling the overpriced H3 is using our codebase, open source, and add his things in a closed source firmware version.
For anything H3 related, we can not answer here as we are not the one doing the closed source H3 thing.

closed 2024-11-02 by gullradriel

324569375, 2024-11-05:
> Please contact your H3 reseller and ask him why he is not pushing back the code modifications he did.
> Because that's what is happening.
> We are not making a firmware for a specific version. We are [...]

Has Mayhem fallen so low without Furrtek and James Shao? They only make minor modifications to existing features, yet they criticize others for not being open-source. They claim that H3 uses their codebase, but conveniently ignore the contributions H3 developers made to Mayhem. If Furrtek and James Shao were still serving Mayhem, one can only imagine how much more successful it would be. The management of Mayhem should also reflect on their own shortcomings in light of the current situation.

gullradriel, 2024-11-05:
Well, we are all sorry but our crystal ball is broken, and we can not help you for code we do not have  X-D

Also, your only activity using that account ( [324569375](https://github.com/324569375) because you have no name, no nick, just a pesky number) was creating that very specific issue, booooh :-1: !

If you gave your money for a H3, then contact the guy who is making the H3 firmware. If ever he is willing to push back a PR here, I'll review it with pleasure and even try to help with his contribution.  Ho, And we never charged / denied his past, open source contributions. 

Now to the technical points of your question: 

- Portapack processing power is based ON the processing power of the HackRF, and only have a 120MHz + a few MB of memory processing power.

- if you want to receive SSTV, using a computer or  a laptop and QSSTV (Linux) or MMSSTV (Windows) is going to provide way better results than using any app on Mayhem. 

- Same for NOAA, use NOAA-APT will provide better results, will be easier to use, than any app in Mayhem, for the same reason mentioned in the first point.

I consider that we have largely answered you, and that you're a bit of a troll (aren't you ?)

:-)

htotoo, 2024-11-05:
Yeah, praise those, who break license to get money from people instead, and not serving the community, just own pocket.
Talented? Yes. Greedy? Of course? Good for anyone else? No.

But if you check the changelog, there was tons of progress here. So don't be rude, and don't forget: everybody works on this fw, do it for free, and on their free time.

If you wanna more progress, then you can contribute too.

eried, 2024-11-21:
This is one of the few cases I would recommend to get an H3 😬

gullradriel, 2024-11-21:
> This is one of the few cases I would recommend to get an H3 😬

Yes, and then go to the reseller's / closed firmware provider repository to complain, instead of here :-)