portapack-mayhem/mayhem-firmware #2336: Incorrect  FFT  in WFM  IF Frequency response , in  App Audio RX  in latest nightly 3rd Nov -

Brumi-2021, 2024-11-03:
### Describe the bug.

Hi,  after several weeks not testing,
today ,  I updated my PP binary  with latest nightly 3rd October 24, 
and I realisded that in the App Audio RX , the FFT of the demodulated FM IF in WFM mode is wrong . 

### Reproduction

It is easy to see,  just , looking  for 19khz stereo tone in a STEREO BROADCASTING FM radio ,now  it is  NOW clearly missing , 
I checked quickly several  previous  versions , and so far , I can confirm , that in nightly   30th Sept (was still  OK), 
 but from 7th Oct (it is NG) .    (Pending to check between) .   

We need to check more carefully which commit  broken it . 

### Expected behavior

I will add two pictures,

gullradriel, 2024-11-03:
It may be this.
I think @zxkmm tried to tweak this.
The revert PR may be in the tube or waiting approval.

Brumi-2021, 2024-11-03:
Hi @gullradriel ,    and @zxkmm 
I checked  nightly 6th of October and it is working fine.  (still perfect)

So the side effect appeared from  nightly 7th of October , 
There are two commits on there 

Revision (n_241007):
https://github.com/portapack-mayhem/mayhem-firmware/commit/83b65ba6ce4d4b8f98b05f48f284168817da9553 - @htotoo: I2C device manager (https://github.com/portapack-mayhem/mayhem-firmware/pull/2282)
https://github.com/portapack-mayhem/mayhem-firmware/commit/d4edb5f5f95b3b69e83a0ff1ca7db378c79cf8e5 - @zxkmm: setting&autostart&widgets (OptionField and Waveform) imp (https://github.com/portapack-mayhem/mayhem-firmware/pull/2286)

I still did not check, but I believed that this strange behaviour  comes from here , 
Commit d4edb5f
 
Cheers,

gullradriel, 2024-11-03:
@Brumi-2021 please try with the fresh build I linked here. It should include the waveform revert. I zipped both .bin a ppfw for your convenience:

[fresh_next_03_11_2024.zip](https://github.com/user-attachments/files/17612107/fresh_next_03_11_2024.zip)

Brumi-2021, 2024-11-03:
Thanks @gullradriel , for been so kind. 
that one , it works well again , 

But pls.  @zxkmm could you explain , how you fixed again ?  I still can see that you changed the sign of the formulas , 

Probably , came originally , from here ,   (you changed the logic , and the signs) 
[image]

But now I can see ,  (here only changed the signs ) 
[image]

Anyway , now it works well again , let's  not break again , pls. 
Cheers.

Brumi-2021, 2024-11-03:
Confirmed , we can close that issue , thanks to both @gullradriel ,  @zxkmm  sommermorgentraum

closed 2024-11-03 by Brumi-2021

zxkmm, 2024-11-03:
Thank you for asking 
So previously before the first pr, the digital waveform is upside down (low level is on the top and high level is on the bottom), my first PR tuned the logic but I didn’t read carefully any tuned the analog waveform logic too.
And second PR fixed the analog logic and kept the digital one.