portapack-mayhem/mayhem-firmware #2338: Add IR Transiver

Hemant7121, 2024-11-04:
### Description of the feature you're suggesting.

Please add GPIO Ir Transiver board

htotoo, 2024-11-04:
IR tx in progress by me

Ademolght, 2024-11-06:
> IR tx in progress by me

amazing you can use this module : 
[image]

htotoo, 2024-11-06:
> > IR tx in progress by me
> 
> amazing you can use this module : 

I'll use ESP32S3, and that's gpio is 3.3V, and not sure this module's data part will work with 3.3. But the implementation will just say: use th GPIO N to send IR. the wiring and module selection will be your choice.

Hemant7121, 2024-11-06:
Everything is set up correctly, but the application also requires the IR sensor to be operational.

tomventa, 2025-06-12:
@htotoo IR TX can be also done using the audio jack, seems like an out of the box solution considering the Jack IR blasters on AliExpress/Amazon