portapack-mayhem/mayhem-firmware #2339: Function of ini-files?

Waycco, 2024-11-05:
### Describe the bug.

When u change ini file frequency for example APRS frequency the APRS app always starts with the US frequency. It would be nice that you could store the initial setting of each app.

### Reproduction

Open file editor change the value of corresponding app ini file for example APRS frequency. Open APRS and you will notice that it does not open in that frequency you stored.

### Expected behavior

The ini setups should stay default or be saved as last used configuration.

gullradriel, 2024-11-05:
I'm doing it for APRS RX, and so far so good, I think it's working.

There is a testable ppwf here: 

https://discord.com/channels/719669764804444213/722101917135798312/1303335520935346186

For any other apps it may also need some enhancement/modifications on our part. 

Please open one issue for each application suffering the problem (some apps are not using app_settings except for LNA/VGA/AMP values)

gullradriel, 2024-11-05:
APRS-RX should be fixed in https://github.com/portapack-mayhem/mayhem-firmware/pull/2340

gullradriel, 2024-11-05:
Some precisions concerning app-settings ini files: 
- You should never edit them by hand, it can cause apps not starting at all
- While a lot of settings are saved by default as receiver/transmitter settings, not all of them are loaded at app startup (that's what you experienced with APRS-RX)
- If you think something should be saved / loaded in app settings, it's always better to open an issue for it
- Please open one issue per app, and one issue per app problem if ever

closed 2024-11-05 by gullradriel

Waycco, 2024-11-08:
Hi, tested the latest nightly build and now
APRS is keeping the set frequency "in memory". Excellent 👍