portapack-mayhem/mayhem-firmware #2342: M10 Radiosonde only partially decoded

he60, 2024-11-06:
### Describe the bug.

I was playing with M10 for a future weather ballon and i seen that it only detect the fact that is meteomodem, gps data and battery voltage but it does not decode anything else, no SN, no Temperature & no Humidity. (btw, the crc check must be disabled, if its activated, it just does not decode anything)
Its properly decoded on sdr++ using RadioSonde decoder

[image]
[image]
[image]

### Reproduction

Go to RadioSonde decoder and try to receive Meteomodem M10

### Expected behavior

show weather data

### Anything else?

I can give you spectrum recording in case, i've tried looking into the code of mayhem & RSdecoder but its just black magic for me ^^''

htotoo, 2024-11-06:
If you can attach the portapack recorded iq file here, that could help for the dev, who will chech this.

he60, 2024-11-06:
Sure ^^
here the IQ & audio, its actually transmitting -25°c, about 38%hum and gps are 90°N - 0°E 150m altitude
[m10.zip](https://github.com/user-attachments/files/17650806/m10.zip)

I hope you can fix that :)

htotoo, 2024-11-07:
maybe could you attach the log for this, that the pp generates? (that may contain your / sonde) coordinates!)

he60, 2024-11-07:
Sure ^^
here it is with _i hope_ false gps data
(90°, 0°)

[logs.txt](https://github.com/user-attachments/files/17666457/logs.txt)

htotoo, 2024-11-19:
Here is a test version. With the provided IQ file the fields changed and there was a lot of errors.

Please test as mutch as you can, and feed back if it is ok, or not. Sadly I can't decide :)

[portapack-mayhem_OCI.ppfw.tar.zip](https://github.com/user-attachments/files/17814424/portapack-mayhem_OCI.ppfw.tar.zip)

he60, 2024-12-27:
CRC looks to be still glitched but temperature & humidity is here thanksss ^^
i would love to learn how you are capable to reverse ing data frames and decode them bc i've never find anything about how to decode them or how they are encoded

htotoo, 2024-12-27:
I'll make a pr soom, so you'll see the modifications.
Crc is out of scope for me in this.

closed 2024-12-28 by gullradriel