portapack-mayhem/mayhem-firmware #2350: White Screen Issue on Mayhem H4M Board with HackRF One

seeklakhan, 2024-11-08:
### What happened?

After connecting the Mayhem H4M board to my HackRF One, the screen remains white on startup and does not proceed to the Mayhem menu. I’ve attempted basic troubleshooting, but the issue persists. I’m seeking guidance on resolving this problem.

Steps to Reproduce:

Connect the Mayhem H4M board securely to the HackRF One.
Power on the HackRF One.
Observe that the screen remains white and does not display the boot logo or main menu.

Tried different combinations of buttons at startup, including long presses, without any response.

### Your hardware

H4M 

### Tests

- [X] I managed to update/boot previously

gullradriel, 2024-11-08:
You managed to boot previously, with the same configuration ?
Are you booting with or without an external power supply plugged in ?

seeklakhan, 2024-11-08:
I was able to successfully boot the H2 PortaPack prior to attempting to boot the H4M board for the first time with the same HackRF One setup. The H2 worked without any issues, but I encountered the white screen issue only after switching to the H4M.

I've tried booting both with and without an external power supply plugged in, but the issue persists in both cases.

seeklakhan, 2024-11-08:
Have you tried to boot without the sdcard in ?
triyed Both with and without SD CARD

gullradriel, 2024-11-08:
Which firmware version are you using ?

seeklakhan, 2024-11-08:
MAYHEM 2.0.2

gullradriel, 2024-11-08:
Well. I would try to boot in DFU, flash latest with dfu_utils.
Also check that the coin battery in the H4M is good, or remove it for a test. A dying coin cell is known to bring problems.

gullradriel, 2024-11-11:
You may try:
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Updating-the-Xilinx-CPLD-on-hackrf-board
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Factory-Defaults

gullradriel, 2024-12-20:
Closed due to lack of activity

closed 2024-12-20 by gullradriel