portapack-mayhem/mayhem-firmware #2368: Splash boot

Kingdamienjl, 2024-11-19:
### Description of the feature you're suggesting.

A setting to randomly display the bmp in the splash folder on boot would be great!

### Anything else?

More apps!

MannyV143, 2024-12-24:
NotherNgineer....sorry....I'm still learning all of this... where do I find it?

htotoo, 2025-09-09:
Without using pmem space, if there is no splash.bmp in the root, just pick randomly from the /SPLASH folder it he Show splash is enabled. Easy to do ;) 
(But please don't do a list of the files in the splash folder, just count the bmp-s, pick a random number, iterate again, and then choose the nth image. This way we don't waste ram, and it won't crash if there are a lot of splash screens.)