portapack-mayhem/mayhem-firmware #2378: Chat application

sn0ren, 2024-11-20:
### Description of the feature you're suggesting.

Suggestion for a new chat application:

We already have an excellent POCSAG receiving application, as well as an excellent POCSAG transmitting application. Combining these into a user friendly texting application to send messages between Portapacks, would be very useful. Maybe with the ability to select various protocols, for instance Flipper Zero also has a Sub-GHz chat application, that it could be compatible with https://github.com/xMasterX/all-the-plugins/tree/dev/apps_source_code/esubghz_chat or it could support APRS messaging as well, or the messager used in the kamilsss NuNu firmware for Quansheng radios https://github.com/kamilsss655/uv-k5-firmware-custom?tab=readme-ov-file 

The application should have a main screen consisting of a chat history displaying  (preferably saved and reloaded to and from a file) and an input field and keyboard below. A separate settings screen where you can set your callsign, protocol and frequency, power output, etc.

Waycco, 2024-11-29:
Excellent Idea, maybe encryption added too to the messages?

strijar, 2025-04-08:
I've been tinkering with the Olivia protocol for firmware my HF transceiver "Brass". I think I might try using the same protocol for chat in Mayhem. RTTY or BPSK are also suitable for these purposes.