portapack-mayhem/mayhem-firmware #2383: out of memory

Willem0810, 2024-11-22:
### Describe the bug.

For example, if I use Recon and go to config and then select input file, I get a message on my screen: M0 Guru Meditation with hint: Out of Memory. I can't figure out what's going wrong here? can anyone help me!
I am using the Porteback H4M Mayem version v2.0.2.

### Reproduction

ja

### Expected behavior

x

gullradriel, 2024-11-22:
Yeah, there is sometime a memory fragmentation problem in some big apps like Recon.

You should try the next nightly (tomorrow's one) in which we got back some memory for our apps.

If you're in a hurry, get the freshly built ppfw file here: https://discord.com/channels/719669764804444213/722101917135798312/1309555918769225800

And flash it using https://hackrf.app , click on manage firmware, flash custom firmware, select the ppfw tar file, wait for your device to reboot.

gullradriel, 2024-11-23:
I tested a lot today, and the bug is not appearing in latest next  version.
Use the nightly, or wait for next stable :-)

closed 2024-11-23 by gullradriel