portapack-mayhem/mayhem-firmware #2387: positioning config menu

f1ghy, 2024-11-24:
### Description of the feature you're suggesting.

The idea would be to have a configuration menu to:
- manually choose your GPS coordinates,
- memorize one or more coordinates,
- memorize the last position find,
- choose the source of GPS coordinates. (ESP32, future to come; Ext NEMEA).
Then, you could only connect the external module if necessary, enter coordinates from an external source... or even manually)
thanks

eried, 2024-11-25:
And how the GPS would be used? 

The problem with a nmea source will be needing background processing to keep it alive consuming a big stream, I am not sure if that will be a good thing, vs for example another solution of the app querying the ESP32 when it needs the position.

For sure there is a need of researching GPS related stuff

eried, 2024-11-25:
(with my question I actually wonder how you envision it appears in other apps and etc, how it is retrieved, where, in which apps, etc)

htotoo, 2024-11-25:
My question is, how would it be beneficial to have the coordinates stored? Yet I don't know any app that would use it for anything. 
For example when you see adsb map, it'll zoom to the airplane. If you store your last coordinates, would it benefit anything?

eried, 2024-11-25:
> My question is, how would it be beneficial to have the coordinates stored? Yet I don't know any app that would use it for anything. For example when you see adsb map, it'll zoom to the airplane. If you store your last coordinates, would it benefit anything?

Well, in that case I can imagine being able to calculate the distance to me, or visualize airplanes around some specific radious, but I also wonder on the use cases (besides adding functionality just for the sake of adding functionality)

gullradriel, 2024-11-25:
The basic functionalities:

- manually set/edit/remove GPS coords with labels
- select a Label as a position and set it for apps using GPS

Can be useful for war-driving in known positions without have an ESP32 attached

f1ghy, 2024-11-25:
Hi, I think that:
- Store manual may be usefull for people who don't have GPS, to try GPS fonction (yess it may be fine for ADSB,  APRS, or FOX HUNT to center the map on your position, yess it does if there is GPS coordonate 😉). ( whach that the maps does't locate you on 0⁰, 0⁰ when the coordonate are null ?)
- Store the las position make you able to unplug the GPS when your in a fixed position (I have the case where when you plug USB (OTG), RF noise make poor reception).